bigtreetech/BTT-TFT35-E3-V3.0 #2: can't use SD Card on LCD from marlin mode, but can use it from touchscreen mode

robrogers, 2020-01-27:
can't use SD Card on LCD from marlin mode, but can use it from touchscreen mode

hfog, 2020-01-30:
This seems like expected behavior. The Marlin mode is really just treating the TFT35 as a dumb replacement for the stock E3 screen so it doesn't have any access to extended features offered by the TFT. When in touch screen mode the TFT is really just a second device hooked up to the printer over serial, think just like Octoprint or Pronterface. So the touch menus are interacting with the TFT which in turn is driving the printer board over serial while the Marlin menus are interacting directly with the printer board.

thisiskeithb, 2020-04-05:
Make sure you set `SDCARD_CONNECTION` to `LCD`:

https://github.com/MarlinFirmware/Marlin/blob/bd4e38860b0ceacd543df8d6cc03a252028efd56/Marlin/Configuration_adv.h#L1200-L1209

```cpp
/**
   * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
#define SDCARD_CONNECTION LCD
```

Mowhok, 2020-05-27:
I have the same issue just in reverse. in marlin mode i can go to sd card but in touch mode i get a unknwon command xyucal1 gco 

with any sdcard added

oldman4U, 2020-06-21:
@bigtreetech Could you please close this unused ticket

This could help new users find tickets which already cover their topic.

Thank you

fineobjetcs, 2020-09-07:
I have same troubles. 
I cant see SD in marlin mode and see it in BTT mode when i choose LSD in #define SDCARD_CONNECTION
But if i choose ONBOARD i see onboard SD in marlin mode and cant see it in BTT

oldman4U, 2020-09-08:
In Marlin you can not see anything TFT related. In touch mode you can see the TFT and you can define if you want to see the sd card from the board or not in config.ini. For this the board sd card must be enabled in Marlin board firmware.

Portzal, 2020-09-08:
Just to throw in my recent experience, I hooked up a new TFT70 and was unable to print from the SD card in Marlin mode, I switched to touch screen mode and was able to access the files on the SD card. 
Without changing the firmware on the board (SKR V1.4 Turbo) and connecting a TFT25 with hasn't had firmware update in over 8 months I was able to access the files on the SD card in Marlin mode.  
I have not tried updating the firmware on the TFT70, and will not be able to for the next 2 weeks. 
My opinion is it is a screen firmware issue. 
Old Screen with old firmware works as expected, new screen with new firmware doesn't.

oldman4U, 2020-09-19:
@Portzal 

SD card of mainboard or TFT?

Portzal, 2020-09-19:
@oldman4U screen.

oldman4U, 2020-09-19:
Marlin can only see the sd card from the mainboard and not the sd card and the U disk from TFT. This is independent of the screen size as long as it is a TFT from BTT.

Portzal, 2020-09-19:
@oldman4U I have a BTT TFT25 and a BTT TFT35 that both print from the SD card in Marlin simulation mode. The new TFT70 does not, so either there has been a hardware change with the new screen or it is a screen firmware issue.

oldman4U, 2020-09-19:
But this has nothing to do with this ticket.

To be able to print from mainboard SD card using Marlin simulation the SD card functionality has to be enabled in the firmware of the mainboard (default OFF) and the TFT (default auto-detect - so off in case it is off in Marlin fw)

Hope this helps

[image]

[image]

Portzal, 2020-09-19:
Forgot anything about the main board because I have not been referring to it. And from the way I read the original ticket neither is that. 
All my comments are about the BTT TFT screens and accessing the SD card inserted in the screen SD slot. Same as the originator of the ticket. 
With a SD card inserted in the BTT TFT70 I can access print files in the touch screen mode, if I switch to Marlin simulation mode on the TFT70 it does not show print files. But for example on my BTT TFT35 V3.0 with a SD card inserted in the screen SD slot I can access the files on the SD card and start a print.

oldman4U, 2020-09-19:
Not sure we can leave the mainboard completely out of the game.

There is a setting in configuration_adv.h, which allows you to define which SD Card Marlin can see:
  _/**
   * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  #define SDCARD_CONNECTION LCD_

On my machine I use a SKR E3 DIP which is connected with a single EXP connector, and on this machine defining   #define SDCARD_CONNECTION LCD does not make any difference.

I could imagine, that using a mainboard with 2 EXP connectors could eventually work, but this is completely guesswork from my side.

So far I was never able to see the TFTs U Disk or SD card from the Marlin emulator, but only when using the Touchscreen mode. But I got a SKR v1.4 last week which I bought for testing. Maybe I can find the time to test it.

So when you use the SD Card of the TFT, does this look the same in Marlin like when you use the on board SD or different?

Portzal, 2020-09-19:
EXP1 and EXP2 is connected to the SKR V1.4 Turbo from the BTT TFT35 V3.0, with a SD card inserted in the BTT TFT35 V3.0 SD card slot and With the BTT TFT35 V3.0 in Marlin simulation mode I can see the files on the SD card then select and print them. I have been using my screen in this way for over 8 months as I prefer Marlin simulation mode to the touch screen. 
No SD card is in the main board,

Portzal, 2020-09-19:
> Not sure we can leave the mainboard completely out of the game.
>
>
>
> There is a setting in configuration_adv.h, which allows you to define which SD Card Marlin can see:
>
>   _/**
>
>    * Set this option [...]

I have not changed any of these settings in my Marlin build.

oldman4U, 2020-09-19:
It is a miracle then...

Portzal, 2020-09-19:
@oldman4U it's not a miracle it's just how it works. The inclusion of your comments in this thread have not been of any assistance as you are basing them on a setup that doesn't allow printing from the SD card in Marlin simulation mode. 
The fact is that you can access files from the SD Card inserted in the TFT screen slot in Marlin simulation mode when you have EXP1 and EXP2 connected to the main board.

oldman4U, 2020-09-20:
@radek8

Sorry to bother you but you know the hard and software much better than me. 

What is needed to be able to print from TFT U disk and SD card, using MARLIN emulation mode?

Does this work on displays with just one EXP connection or does this only work with two?

Whats has to be activated/selected software wise?

Does this work on the new TFT70?

You help is appreciated.

THANK YOU

Portzal, 2020-09-20:
As far as I know only SD card is supported in Marlin simulation mode, and again to my knowledge you need EXP1 and EXP2 cable's connected. I am back from holiday and can now do some fault finding of my own. I will flash the latest.bin from the Bigtreetech screen firmware to my TFT70 and see if the issue has been sorted. If that doesn't work I am going to load a previous firmware version that I have on my TFT35 V3.0 onto my new TFT70 and see if that resolves the issue.
Depending on what I find I will report back here and do a new bug report if needed.

oldman4U, 2020-09-20:
Corrected my question to radek8.

Not sure the TFT70 will support older firmware than it has been released with. 😬

radek8, 2020-09-20:
oldman4U to see SD card on TFT you have to meet 2 conditions.
1. You must be connected via EXP1 and EXP2 (+ correctly declared display type and pins of the EXP2 connector in Marlin). You write that you use EXP3, so you won't see SD on TFT.
2. You must be configured in Marlin
# define SDCARD_CONNECTION LCD

As for the TFT70, I have it. He doesn't know if there are any changes on the TFT70 compared to the TFT35.

radek8, 2020-09-20:
I would not record older FW due to SD support.
A better solution would be to lead the pins from the EXP2 connector (on the Marlin board) to an external SD card slot.

radek8, 2020-09-20:
for example, this
https://www.aliexpress.com/item/32813538365.html?spm=a2g0o.productlist.0.0.6f83443aItqWCY&algo_pvid=2d2c5e73-59fd-4d67-99c7-c8cea8132f53&algo_expid=2d2c5e73-59fd-4d67-99c7-c8cea8132f53-17&btsid=0b0a0ae216005941330491088e3213&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Portzal, 2020-09-20:
I'm out

fineobjetcs, 2020-09-20:
Hi guys !

I find time for some testing. I removed my BTT 35 TFT from one printer and
installed it on another one with the same SKR1.4 and working display. I
make booth connection - rs232-tft and exp1 exp2

But i have not happened
LCD SD in Marlin mode still not worked

I tried onboard SD and it worked fine. I think it's BTT mistake because i
try to change all settings in Marlin and it doesn't matter

My mistake - that i bought this screen 👎because i can't use SD and U disk
in base (Marlin) mode, also i don't understand - for why Marlin mode on
this screen*,* without SD and U disk. Only for fan ?

radek8, 2020-09-20:
SKR 1.4 has its own card slot and the TFT display also has its own card slot and as a bonus it has USB.
Isn't that enough for you?
Need to print from two cards at the same time?

fineobjetcs, 2020-09-20:
Its enough for me while it will be work. But now not. I want to print
from one of two LCD slot SD or USB in MARLIN Mode.
Nobody cant say me how

oldman4U, 2020-09-20:
OK, I also found some time for testing and it works using the SKR v1.4 and the TFT35 E3 v3. Using this hardware I can print from all 3 options using Touchscreenmode and using Marlin I can choose to print from mainboard SD card OR TFT SD card slot depending on the settings I defined.

@Portzal: Why have you deleted your comments? I wanted to thank you, because you have been right and I had the chance to learn something new - very cool! Thank you

@fineobjetcs Which version of the firmware are you using on your TFT? Can you send me the config.ini of it to check. Thank you

fineobjetcs, 2020-09-20:
@oldman4U

I try last firmware for screen from BTT github/ and Marlin 2.0.5.3
Its last config but i try many changes for test

oldman4U, 2020-09-20:
Marlin is currently 2.0.6.1 and seems to be OK.

TFT is called "master" and is currently 4 days old. There seems to be a problem with the precompiled versions but you will compile your own firmware anyway I guess;-)

radek8, 2020-09-20:
So once again.
Marli only supports a single SD card!

when setting Marlin to
#define SDCARD_CONNECTION ONBOARD
There will be an SD card on board with Marlin available in Marlin.
An SD card connected via EXP2 on the display (even by touch TFT) in Marlin will not be available
Both cards + UDB disk will be available in touch mode.

When setting Marlin to
#define SDCARD_CONNECTION LCD
The SD card on the Marlin board will be unavailable for both Marlin and touch mode.
The SD card on the display connected via EXP2 will be available in Marlin and in touch mode.
USB drive is only available in touch mode.

Is my English clear enough?

oldman4U, 2020-09-20:
To me everything is clear and I checked on the SKR v1.4 which I bought 2 weeks ago. The E3 DIP and Pro which i use work differently. Thank you

radek8, 2020-09-20:
SKR E3 DIP has port EXP3, so you can't get to the SD card of the display from Marlin.

radek8, 2020-09-20:
Of course, the settings of the display driver depend on this

Enable the driver for cables EXP1 and EXP2
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
with it, the SD card pins are activated

You enable the driver for the EXP3 cable
#define CR10_STOCKDISPLAY

oldman4U, 2020-09-20:
Again. Thank you for your help. I really appreciate.

oldman4U, 2020-09-20:
> SKR E3 DIP has port EXP3, so you can't get to the SD card of the display from Marlin.

Exactly. And the SKR Pro v1.1 and SD card support is a special chapter....🤪

oldman4U, 2020-09-20:
> Of course, the settings of the display driver depend on this
> 
> 
> 
> Enable the driver for cables EXP1 and EXP2.
> 
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> 
> with it, the SD card pins are activated
> 
> 
> 
> You enable the driver for the EXP3 cable.
> 
> #define CR10_STOCKDISPLAY

Traffic light added this information to his READ ME BEFORE STARTING A TICKET because i asked him to do do. Maybe he can also add the information about the Sd support depending on the availability of EXP2.

Portzal, 2020-09-20:
> OK, I also found some time for testing and it works using the SKR v1.4 and the TFT35 E3 v3. Using this hardware I can print from all 3 options using Touchscreenmode and using Marlin I can choose to [...]

@oldman4U I deleted my comments from this thread because what I was saying was not being understood.
So I assume it would be pointless to have my comments here.

oldman4U, 2020-09-21:
@Portzal 

But your comments pushed me into the right directions. So I am happy you are back. Sorry that I did not understand immediately. 

👍🏻😎

fineobjetcs, 2020-09-21:
I have important news! I change screen from 3.5 to new 2.5 and SD start
working in marlin! With same settings in skr board. Thats troubles with 35
screen only. May be firmware or etc. 25 screen works with sd right and in
btt mode I see 3 medias: sd LCD, sd on board and u drive!

oldman4U, 2020-09-21:
Using new TFT firmware this also works with TFT35. New TFT already comes with new firmware. I see all 3 in touchmode and one sd card in Marlin mode on my TFT35. 

Let me know in case you need help updating the TFT.

RedCoder09, 2020-09-21:
So... find a solution yet? If so, what is it?

oldman4U, 2020-09-22:
Are you robrogers or are you new? What is your mainboard and tft and which firmware are you using?

oldman4U, 2020-09-22:
The solution is the latest firmware an the correct setup.

Portzal, 2020-09-22:
I have updated my TFT70 V3.0 to latest firmware, still no good. I then compiled my own build altering settings that I thought may be interfering and created a .bin. I still get SD fail in Marlin simulation mode. I plug in my old TFT25 and no problem I can print from the SD card inserted in the screen in Marlin simulation mode. There is a bug with the new firmware.

oldman4U, 2020-09-22:
Can you please attach your config.ini.

Thank you

Portzal, 2020-09-22:
I am beginning to think it might be a hardware issue with my screen. I purchased it as soon as it was released and maybe poor quality. I have toggles all the relevant settings in config.ini, so I doubt me sharing it will help. I have tried it on a different board with exactly the same results.

oldman4U, 2020-09-22:
Feel free to send me the config.ini or any other information which could help us to help you.

For Tony this worked very well today: https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/81

Portzal, 2020-09-22:
I am going to cross reference the screen firmware .h files for the version I have on my TFT35 V3.0 with the latest version and see if I can pick anything up that may be causing the issue when I have a spare moment. 
As far as hardware possibilities it may be a bad pin in EXP1 or EXP2. But I would have to research the pins.
Seems like a good opportunity to start using the touch screen mode.

oldman4U, 2020-09-23:
Ok. Maybe you can start your own ticket then and let us know👍🏻

@robrogers

Please help the community and close the ticket once you do not need it anymore. Thank you

RedCoder09, 2020-09-24:
TFT35, SKR E3 Mini V1.2, and I'm not robrogers.

oldman4U, 2020-09-25:
Nice to meet you and nice hardware.

oldman4U, 2020-09-25:
Please help the community and close the ticket once you do not need it anymore. Thank you

xdased14x, 2020-11-08:
i know this is an old ticket just checking that i understand.
i have ender 3 pro with skr mini e3 1.2, on marlin 2.03(will be updating to 2.0.7 shortly) i added tft 35 e3 v3 to it i can see sd card (on board only in sim mode) in touch i only see tft sd and u disk i want to use on baord in both so i uncomment this   #define SDCARD_CONNECTION LCD in marlin config adv h(line 1166 on vscode)

radek8, 2020-11-08:
line must be changed to
  #define SDCARD_CONNECTION ONBOARD

xdased14x, 2020-11-09:
ok i'll try that thanks
so from lcd to onboard

mark429, 2021-01-21:
> So once again.
> Marli only supports a single SD card!
> 
> when setting Marlin to
> #define SDCARD_CONNECTION ONBOARD
> There will be an SD card on board with Marlin available in Marlin.
> An SD card connected via EXP2 on the display (even by touch TFT) in Marlin will not be available
> Both cards + UDB disk will be available in touch mode.
> 
> When setting Marlin to
> #define SDCARD_CONNECTION LCD
> The SD card on the Marlin board will be unavailable for both Marlin and touch mode.
> The SD card on the display connected via EXP2 will be available in Marlin and in touch mode.
> USB drive is only available in touch mode.
> 
> When connecting the display only via EXP3
> the SD card in the display will not be available in Marlin with any settings.
> 
> Is my English clear enough?

Thank you for this. I didn't understand why it would work from SKR1.3 but not SKR Mini1.2,  but now it makes sense. 
Thank you!

radek8, 2021-01-21:
similar query
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1542