bigtreetech/BTT-TFT35-E3-V3.0 #3: Continuous beep when switching back to the touch screen

94Porsche968, 2020-01-30:
Switching between the screens works fine but when switching back to the touch screen the machine beeps continuously.  The mainboard is an SKR E3 Dip-V1.1.  I've only been able to stop the beeping by powering down the machine.  The touch screen is running with BIGTREE_TFT35_V3.0.25.1.

hfog, 2020-01-30:
I have encountered exactly the same thing. I can silence the beep by turning off the speaker in the machine settings menus but that turns off all beeps on the device so it's not an ideal solution. 
I came here specifically to register this same issue so I'm glad I'm not the only one to hit it.

critter42, 2020-01-31:
I am experiencing the same issue. It's quiet on boot-up, then I switch to Marlin Mode and it is fine also. I switch back to Touch Screen and it starts the tone. It will stop if I switch back to Marlin, but then start again when I go back to Touch Screen. I don't do ABL so I can't advise on the other issue

oldman4U, 2020-07-05:
Hi @94Porsche968.

Seems this issue has been fixed for sometime now. So could you please close this ticket.

Thank you

closed 2020-07-07 by 94Porsche968

Next-victim, 2020-11-20:
Just dealt with something like this. My solution was to comment out PID_DEBUG with both PIDTEMP and PIDTEMP active. I had enabled PIDTEMPBED and the debug right before this started happening, no consistant beeping after commenting the debug out.

Board: SKR 1.4 turbo (24v setup)
Screen: BTT TFT43
Attached USB: Pi4 running current OctoPi