bigtreetech/BTT-TFT35-E3-V3.0 #6: Encoder Wheel LED_ORANGE defined wrong

critter42, 2020-01-31:
when you select "Orange" for the Encoder Wheel LED, it is actually Green.

Looking in src\User\Menu\ledcolor.h  you have:

#define LED_ORANGE                0x00FF8C00

Since  you are defining the color in GRB order rather than RGB order, it should read

#define LED_ORANGE                0x008CFF00

However I find that 

#define LED_ORANGE                0x0045FF00

Is much closer with these LEDs to what I think of as orange. 0x008CFF00 is still way too heavy on the green - the green portion of these LEDs really overpowers the other colors.

thisiskeithb, 2020-04-03:
Thanks for the report!

I've put in a PR to fix both Orange & Violet LED colors, so be sure to grab firmware from the main TFT firmware repo once this is merged: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/527

>  the green portion of these LEDs really overpowers the other colors.

I agree. Even pure white (`0x00FFFFFF`) has a green tinge to it.

oldman4U, 2020-06-12:
Hello critter42.

Have you tried the latest Master Branch which should solve the issue you have reported.

Please let us know

Thank you

critter42, 2020-06-12:
No, as I had already fixed it in my code.
Danny

thisiskeithb, 2020-06-12:
Please close the issue if your problem is resolved.

closed 2020-06-12 by critter42