bigtreetech/BTT-TFT35-E3-V3.0 #9: Permanent blank screen in Marlin Mode on SKR 1.3

Andy-ABTec, 2020-02-08:
Hi,
Just purchase a BTT TTF35-E3 V3.0 to use on my Ender 3 along with an existing SKR 1.3.

Touch Mode appears to be fine but Marlin Mode just presents a blank screen. All firmware is the latest available, Marlin 2.03 for the SKR and BIGTREE_TFT35_V3.0.25.1 for the TFT, and the boards are interconnected with all three cables (correctly orientated).

The SKR is using BTT2209 V1.2 driver boards in UART Mode.

Happy to provide whatever else you need.

Also noticed that the  mode switch screen setup seems counterintuative, in that turning the knob clockwise (top to the right) moves the selection to the left and vice versa.

Thanks in advance for a timely response

Andy B.

hfog, 2020-02-08:
After you go to Marlin mode, try turning the knob. It often presents a blank screen until you turn the knob, causing the screen to refresh.

Andy-ABTec, 2020-02-08:
Been there - done that - nothing...

Andy-ABTec, 2020-02-10:
Having look further this does not appear to be a firmware problem, I have now tried:
```
BIGTREE_TFT35_V3.0.25.1.bin
BIGTREE_TFT35_V3.0.25.bin
BIGTREE_TFT35_V3.0.24.2.bin
```
From the GitHub Repository and they all have the same problem - the screen is TOTALLY blank in Marlin Mode...

testproggie, 2020-02-10:
After having struggled greatly (noobie here) I am at the same point although I have SKR 1.4 Turbo. Things work through the BTT menus, but on the Marlin side it is just a blank screen. I'm am not certain at this point whether I even need to use the Marlin side, but it would be nice to look through everything. Still working on getting other stuff right in the config files.

Andy-ABTec, 2020-02-11:
So, I just built my own firmware for the screen...
I made a small change to Configuration.h, removing the comment mark from the Banner Text definition viz:

`   #define` ST7920_BANNER_TEXT "LCD12864 Simulator"

And now when I switch to Marlin Mode I see the banner but the rest of the screen is still blank and turning or pressing the encoder still has no effect. So the actual screen is functional but not working as advertised...

testproggie, 2020-02-14:
Wow, being a newbie in this stuff is tough. I still haven't found a good guide that explains all the functions of Marlin. Lots of youtube videos, but hunting through videos to see if they address your particular problem is time consuming. As to the issue above, my issue is that in the Marlin configuration.h file there is no LCD defined by default. For me uncommenting "//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" was the trick. Hopefully somebody will stumble across the and be able to make use of it  :)

Andy-ABTec, 2020-02-14:
I know, found that a while ago, fixed the "No printer attached" message in Touch Mode...

Andy-ABTec, 2020-02-17:
For what it's worth... **ITS WORKING!** - and I think its a Marlin issue...

So, as a last resort while I still have some hair I setup the SKR 1.3 and the TFT35-E3 V3.0 on the bench with four BTT TMC2209 V1.2's interconnecting EXP1 to EXP1, EXP2 to EXP2 and TFT to RS232, and powered by the USB connection on  the SKR.

Firmware on the TFT is 25.1 with the Unified Menu Material theme.
Firmware on the SKR 1.3 is **Bugfix 2.0x** commit 75e3aa1.

I ignored the "Ender 3" configuration and manually made the following changes:

platformio.ini
```
default_envs` = LPC176
```
Configuration.h
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200
#define MOTHERBOARD BOARD_BTT_SKR_V1_3
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```
Powered up, switched to Marlin Mode (blank screen) and clicked again - and Voila!

This has **NEVER** happened before and its not for the want of trying...

While I've got everything on the bench I'll fiddle about a bit more and see if I can isolate the actual problem, but for now at least I'm a happy bunny!

KennyD42, 2020-02-22:
I'm noticing you have set your

platformio.ini
default_envs = LPC176

Is that the proper designation for the SKR 1.3? Mine is set at LPC1768. Could this be my problem??

KennyD42, 2020-02-23:
> For what it's worth... **ITS WORKING!** - and I think its a Marlin issue...
> 
> So, as a last resort while I still have some hair I setup the SKR 1.3 and the TFT35-E3 V3.0 on the bench with four BTT TMC2209 V1.2's, interconnecting EXP1 to EXP1, EXP2 to EXP2 and TFT to RS232, and powered by the USB connection on the SKR (with no messy cable twisting).
> 
> Firmware on the TFT is 25.1 with the Unified Menu Material theme.
> Firmware on the SKR 1.3 is **bugfix-2.0x** commit 75e3aa1.
> 
> I ignored the "Ender 3" configuration and manually made the following changes:
> 
> platformio.ini
> 
> ```
> default_envs = LPC176
> ```
> 
> Configuration.h
> 
> ```
> #define SERIAL_PORT -1
> #define SERIAL_PORT_2 0
> #define BAUDRATE 115200
> #define MOTHERBOARD BOARD_BTT_SKR_V1_3
> #define X_DRIVER_TYPE  TMC2209
> #define Y_DRIVER_TYPE  TMC2209
> #define Z_DRIVER_TYPE  TMC2209
> #define E0_DRIVER_TYPE TMC2209
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> ```
> 
> Powered up, switched to Marlin Mode (blank screen) and clicked again - and Voila!
> 
> This has **NEVER** happened before and its not for the want of trying...
> 
> While I've got everything on the bench I'll fiddle about a bit more and see if I can isolate the actual problem, but for now at least I'm a happy bunny!

I noticed your default_envs = LPC176 for the SKR 1.3. Mine is set to LPC1768. Could this be my problem?

Andy-ABTec, 2020-02-23:
Absolutely not - it's my bad copy-pasting!

closed 2020-02-24 by Andy-ABTec

KennyD42, 2020-02-24:
I've tried to solution that you've typed in and it does not work

steinboy, 2020-03-04:
I had the same issue with "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER"  uncommented but then I commented out #define CR10_STOCKDISPLAY and then it worked!

After first switching to the Marlin view though the screen is black until you move the knob.
btw. you can change the background colour and text colour of the Marlin display in the touch display settings for screen.

Andy-ABTec, 2020-03-04:
my "#define CR10_STOCKDISPLAY" was already commented out but Marlin bugfix-2.0x seemed to resolve the problem (https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/9#issuecomment-586908632), however I never managed to get to the bottom of the problem...
The knob twist thing is normal, but mine wouldn't display at all even thought I could get the Simulator message to display in Marlin mode.

ijwoo1, 2020-03-04:
Thank you!

This had me scratching my head. 

I found Baudrate of 250000 seems to sync faster than 115200, so the key changes I had to make in configuration.h for an Ender 5 with SKR 1.3 were:

//#define CR10_STOCKDISPLAY
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define BAUDRATE 250000

Points to note: you will get "No printer attached" for about 30s after switching on the printer. You may need to go to Graphic mode, menu, settings and correctly set the baudrate but from then on it should connect after a small delay, without needing to change anything.

When you switch to Marlin mode, by pushing in the encoder for 3s, you will have a blank screen until you refresh the display by rotating the encoder.

dmurray01, 2020-04-13:
hardware; 
1. only use one cable which you connect to exp1 on board and exp 3 on display
2. connect tft cable to board other than reset pin, connect other side to display normally

software;
1. uncomment #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

merinkumar, 2020-07-09:
I too had this issue today. But now am able to connect the TFT35V3 in both ways 2 EXP cables and 1 EXp cables. In both case the TFT black cable was connected to my SKR 1.3 board.

One EXP cable:
#define CR10_STOCKDISPLAY
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define BAUDRATE 250000
connect EXP3 on TFT35V3 to EXP1 on SKE1.3 board
connect black TFT cable (Note : dont connect the REST PIN to SKR1.3)

Two EXP cable:
//#define CR10_STOCKDISPLAY
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define BAUDRATE 250000
EXP1 on SKR to EXP1 on TFT35
EXP2 on SKR to EXP2 on TFT35
TFT black cable connected as usual.

if it helps any one.

thanks

NiiXXiiN, 2020-12-27:
> I too had this issue today. But now am able to connect the TFT35V3 in both ways 2 EXP cables and 1 EXp cables. In both case the TFT black cable was connected to my SKR 1.3 board.
>
> One EXP cable: [...]

I just made an account to say thank you, it was that reset pin that made all the difference!

bryanhunwardsen, 2021-02-14:
I just want to add an alternate solution for anyone facing this issue where the listed solution was not the issue.

When switching to emulation mode the screen does not load:
A simple issue I faced and resolved, the marling cfg.h did not match the TFT cfg.ini for the (simple) screen type.
Marlin was set to LCD2004 TFT was LCD12864, flashing either/both to unify to the same screen type corrected the issue.
In my case I re-flashed the TFT to LCD2004 as no recompile required and LCD2004 is a better utilization of the TFT35 screen realestate.