bigtreetech/BTT-TFT35-E3-V3.0 #10: continuous pauses on SKR Mini  E3 v1.2 with TFT35 E3 v3.0

nachosan100, 2020-02-16:
Hello guys, 
i get continuous small pauses in the print when the min e3 and the tft are connected. It seems like the tft is causing the board to wait for processing and thus making the print pause for 2-3 seconds. Posted this on FB and got a repsonse to try if the error persists on the original display. And Voilá, the error is gone and its printing smoothely as always.

I am running customized marlin 2.0.0 with ABL and the SKR Mini E3 v1.2 plus the tft35 E3 v3.0 display.

Any suggestions as to why the rs232 is causing the pauses?

thanx,
Alex

hfog, 2020-02-16:
Does it still happen if you are on Marlin mode?

nachosan100, 2020-02-16:
Yes. I was testing in Marlin mode.

hfog, 2020-02-16:
Try disconnecting just the serial TFT cable. I'd be very interested to know if that helps.

nachosan100, 2020-02-16:
I’ll try that once my print is done. That will be tomorrow. I anticipate this also working as the original display is now making the leintet work flawlessly! Will get back to you. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nachosan100, 2020-02-16:
Was able to test now with the serial cable detached on the tft35 - and it works! So it’s def an issue with the serial communication!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Scope666, 2020-02-16:
I asked @guruathwal about this previously and he said there are no communications happening over that serial line in Marlin Mode, but from the behavior it certainly seems like there is.  It has to be "leaking" in somehow...

hfog, 2020-02-16:
I've been spending a lot of time in the firmware recently trying to get Octoprint to work when connected to one of the TFT UART ports (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/281). I have not found anywhere in the official firmware where it's communicating over the serial port when in Marlin mode but the fact that @nachosan100 found it makes a difference indicates that it probably is doing something there. Frankly, that's why I made the suggestion. I suspected that it was happening and I was just unable to find it. This will encourage me to keep looking for it.

Scope666, 2020-02-16:
@hfog  It HAS to be.  Myself and another guy on Facebook were just discussing how when the TFT is connected, prints fail all the time and / or stutter, even in Marlin Mode.  As soon as we both connected our stock displays ALL the problems went away.  If it's not communicating in Marlin Mode, than the only other possibility is some bizarre electrical interference caused by the TFT.

steinboy, 2020-05-05:
I have the TFT35-E3-V3.0.26 connected to an SKR V1.3 running Marlin 2.0 and it seems that there is some interference on the serial port when running it in touch mode. I made a test print that just makes a lot of fast continuous circles and it causes a short pause approximately every 2 seconds.
This happens whenever the TFT serial line is connected and screen is touch mode, even when printing from the onboard SD card over the USB port from a PC. As soon as it is switched to Marlin mode it is normal again. I tried with different print speeds and the pause interval seems to stay the same, around 2 seconds.  I also have the TFT28 with same version .26 and it does not have this issue. It seems like it is continuously transmitting some query in touch mode even when it is not printing from the touch controller.

steinboy, 2020-05-05:
I made some tests by printing from the onboard SD via USB cable to PC running Pronterface to start the print, and TFT35 in touch mode. While it was running I went into the Settings/Feature and  changed the Filament sensor from SMART to OFF and then suddenly it was running normal without the pauses. When I tried to reproduce the issue though it wasn't consistent. It seemed like it was somehow related to going into that menu and changing something but not specifically the Filament runout. However, once it was running smoothly it would consistently run OK from power up if I didn't change anything in the Features menu.

mysticprysm, 2020-05-31:
Sort of unrelated but maybe can provide additional insight, I recently got a e3 mini v2 and a tft35 e3 v3, and they both works together. However the first two or 3 days I was setting up my board, I found that with bltouch, the TFT35, and the new board, it just wasn't wanting to work completely properly. 

So what I had was, without bltouch, everything seemed to work fine with basic firmware. Adding bltouch, I had various issues like not homing, x thinking it was triggered, and x slamming into the right with the belt skipping a bunch and I of course switched power ASAP. The reason I'm mentioning this, is because once I unplugged my TFT35 e3 v3, everything seemed to work. While I haven't tested a bunch, it showed no initial signs of failure. 

Just a theory, the controls being sent from the screen to the board, it's getting hung up somewhere. No matter what the command may be. Because when it did work on non bltouch firmware, I would go to hit stop print, and it would take maybe a layer or two worth of printing most of the time to stop. So i was thinking it's communicating like octopi just in that it reads and sends commands. I'm not sure the bitty gritty of all of this as I'm zero experience in firmware editing or compiling, except for the brief tinkering I did over the weekend. Hopefully this could be useful to anybody and hopefully I can get some answers because I really like the screen.

guruathwal, 2020-06-01:
This problem was solved in Marlin firmware https://github.com/MarlinFirmware/Marlin/pull/17032
You need to update your marlin firmware to lated version or bug fix branch

mysticprysm, 2020-06-01:
The problems I'm describing have not been fixed. I've tried many different firmwares. I'm also in contact with BTT diretly.

terghalin, 2020-06-16:
> The problems I'm describing have not been fixed. I've tried many different firmwares. I'm also in contact with BTT diretly.

What is your setup? RPi revision and pins that connected to TFT? May be this can help: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/281#issuecomment-644824150

Also, you could try to compile TFT firmware with SERIAL_ALWAYS_ON and RAPID_SERIAL_COMM enabled.

I've sucessfully tinkered my RPi 3 B+ , TFT3.5 v3.0 (UART4) with SKR 1.4.
Connection chain: RPi 3 connected to TFT UART4 via GPIO 14 and 15, TFT connected to SKR 1.4. with EXP (2 cables in my case) and serial TFT.

mikebryniarski, 2020-08-13:
I am seeing a similar thing in my setup (Ender 3 Pro with SKR Mini E3 V2 and TFT35 E3 and octoprint connecting to the USB port on the printer)

If I print something via octoprint I will see random 2-3 second pauses as long as the TFT35 is connected from RS232 to the tft port on the main board. If I disconnect this cable, the problem goes away. I am running a custom build of Marlin based off the 2.0.6 tag

oldman4U, 2020-10-01:
What is the status of the reported issue. 

@mysticprysm What is the outcome of the contact with BTT?

oldman4U, 2020-12-12:
The current master TFT firmware comes with some changes how buffers are used. I would give it a try to see if it solves the reported problem.

graza, 2021-03-08:
As a plus-one to this issue, I came across this issue in github after having the same problem and searching for a cause.  I have just installed the bundle of an SKR mini E3 v2.0 plus a TFT35 E3.  I use Octoprint to send gcode to my printer.  Before I printed anything I updated everything to the latest BTT released firmware (mainboard and TFT), so everything is up to date as of today.  The pauses happen quite regularly.

gregshawdj, 2021-04-09:
I have been experiencing the same issue with stuttering prints I am using a SKR E3 mini V2 and a BTT TFT35 E3 V3.0. I have tried the PSU, Main Boards, Firmwares, Slicers, removing Octoprint, Printing from different SD cards, moving the printer to different rooms, buildings, removing surge protectors the lot.

The issue was eventually found  to be cause by interference to the BTT TFT35 E3 V3.0. I had a SunLu fialdryer next the to the TFT screen as soon as I turned the filadryer off or moved it away from the TFT screen the print stuttering stopped instantly. I have made a video of this happening. https://youtu.be/ohipSSizLVY

makertronic, 2021-05-30:
Hello, i'm habing this issue when i load or unload filament in GUI mode. I'm using "malin mode", cause it's working.

I'm having a second issue, in GUI mode. When i start print from SD, the head do the lvling, then she goes on 0,0 and she extrude, then move Z a little, then extrude, then Z .... no issue in "marlin mode"

patrickschaerer, 2023-01-06:
I've got the same issue with TFT-Firmware E3.27.x and Marlin bugfix-2.0.x
When printing in Touchscreen Mode from sdcard of the display the print pauses from time to time for about a second.
Is there a fix for that? 
In Marlin Mode there is no sdcard from the tft recognized, so I cannot test if these pauses also occur in Marlin mode.

radek8, 2023-05-22:
Are you interested in testing a possible solution to this problem?
If so, write in this thread.
A programmer is looking for someone with this problem to test the fix.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761#issuecomment-1557196649