bigtreetech/BTT-TFT35-E3-V3.0 #13: [BUG] (Stuck on Select Mode, Touch mode Not working after a minute or so)

fenix8k, 2020-04-10:
I was adjusting my K values for Lin_Advance, it was all ok, after the printer finish, I turn it off (after waiting nozzle go under to 60C).
Then I turn it on Again, start checking some configuration from the Touch Screen and it goes out to the Choose mode section (Marlin mode or TouchMode) and there start the nightmare!, After that, I turn the printer Off/On several times, re-flashed the firmware like 20 times with different settings (my screen comes with 25.2 from factory), The firmware update go OKs, an interesting behavior is that if i leave the printer off for like 5 minutes and turn it on, the TouchMode Work OK for like a minute or 2, then it goes back to the mode selection menu.

I try turn it on Only with 5V and GND it don't change the behavior no mater the printer is connected or not, the Marlin mode Work Fine sometimes.

I try to reset the settings on the Screen and it's say it's Ok that i need to powercycle it, but it don't fix the problem, no idea what to do, i was going to burn the bootloader again but i can't find what pins i need to short to put the screen on bootloader load state

It's Posible that a Capacitor is Wrong or something?, it was working fine for line 1 week.

Link to a video where you can see the behavior
TFT35 Bug: https://youtu.be/p25iWA8zpho

[](https://youtu.be/p25iWA8zpho)

Thx!

Expected behavior
TouchScreen Mode Working

Actual behavior
Stuck on Selection Screen, some times Marlin mode Work

Hardware Variant
SKR 1.4 Turbo
TFT35 E3 V3.0
TMC2130
Ender-5

TFT Firmware Version & Main Board Firmware details, uploaded my firmware and Configuration.h if you need it:

Marlin bugfix-2.0.x Last to Date 9/4/2020
[Firmware.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/4463391/Firmware.zip)
[Configuration.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/4463392/Configuration.zip)

Last TFT TouchScreen Firmware 25.3

fenix8k, 2020-04-12:
So after trying allot of things i could figure out what is going wrong, to make it short, the problem is with the digitalizer and the metal case for the Screen on the Ender5, the scree permanently detect a touch, its the same as pressing the wheel constantly, so to solve the problem I just put some tape between the screen and the metal frame and problem solved, I think this is a fail in the design because the TFT Screen is a direct replacement for Ender3/5 Screen.

I think it's important to leave this info in some relevant place so people struggling with this could fix it really Easy

oldman4U, 2020-06-24:
Hi fenix8k.

Maybe you can change the title of this ticket to point to the issue with the PCB touching the metal case instead of a particular issue this can cause. At the moment only customers with this particular issue will pay attention, but the issue itself can cause several other problems.

Thank you

kishkaru, 2020-07-10:
Thanks for pointing this out, I ran into the same issue with my brand new TFT35-E3!

It booted up initially in Marlin mode, and I held down the knob to trigger the Marlin/Touch selection screen. I pressed Touch and nothing happened. I thought I messed up something in my Marlin code, but it turned out to be that the entire right half of the touchscreen wasn't responding to touches! 

I removed the TFT35 from the Ender3 metal display case and the entire screen was working just fine. Then I screwed it back into the display case, and the right half wasn't working again. I think it's because my screen is shifted to the left a little bit (manufacturing defect), and when tightening down the screen to the case, the display ribbon cable gets pinched to the metal frame. I ended up loosening the 4 bolts to the case a little bit, so that the ribbon cable doesn't pinch.

Finally, I used the terminal mode in the Touch mode to verify that every button can be touched/pressed and is triggered accurately.

oldman4U, 2020-07-10:
Hi.

Happy to hear I could help. You can also use some insulation tape to
separate the display from the frame.

Please do not forget to close the ticket once you do not need it anymore.

Thank you

um 22:21:

TimWilde, 2020-07-16:
> So after trying allot of things i could figure out what is going wrong, to make it short, the problem is with the digitalizer and the metal case for the Screen on the Ender5, the scree permanently [...]

Thanks for sharing this; it's just saved hours diagnosing the same problem on another Ender 3. Much appreciated 😄

oldman4U, 2020-08-03:
@fenix8k 

Could you please close this ticket - seems you do not need it anymore.

Thank you

kishkaru, 2020-08-04:
No, I don't think this issue should be closed. This is a hardware design defect with the TFT35 that should be addressed by the manufacturer.
Once someone @bigtreetech acknowledges this issue, they can close it. In the meantime, leaving this issue open will be better for visibility for others running into the same issue.

oldman4U, 2020-08-04:
Then the title should be changed because the missing insulation can cause various issues and not only the one which is mentioned in the headline.

oldman4U, 2020-08-04:
And the display i got 2 weeks ago had the insulation already on it.

oldman4U, 2020-09-07:
@bigtreetech 

Please close this ticket

oldman4U, 2020-09-25:
Please help the community and close the ticket once you do not need it anymore. Thank you

radek8, 2020-09-29:
This is a known issue that users are notified about here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1047

oldman4U, 2020-09-30:
Hi Radek. 

There are so many tickets like this, where users open their first ticket and never reply, because they don't know or don't care...

traffic-light, 2020-10-07:
@bigtreetech please close this ticket.

doug-scott, 2020-12-25:
My TFT35 e3 v3 started doing the same thing the other day, creating the semaphore file in the root of the SD card started the screen calibration function.  Just the fact that I could follow the screen prompts told me this was the solution.  That reset is a bit odd though, I must have gone through it a dozen times before it successfully completed.

t0m541, 2021-02-06:
I wonder if this is what keeps happening to my TFT.
Sometimes it seems to be totally out of calibration so that where I touch is not where the screen thinks I've touched.
A couple of days ago it wouldn't do anything, then managed to get into touch screen calibration screen and even using a stylus on the red dots it was telling me that calibration was unsuccessful.
The next day I switched the printer on and it worked fine even though I power cycled it the previous day to try to clear the fault.

bene3d, 2021-05-19:
Hi! My solution was to loosen the screws a bit. Just one mm. Then it worked again. Thanks for pointing out that it is a hardware issue (short between the case and the pcb),

ericdraken, 2022-06-26:
> I think it's important to leave this info in some relevant place so people struggling with this could fix it really Easy

It helped me. I was able to push up the LCD screen and cause it to move a mm or so and now the select screen goes away. Yes, tape on the edges of the metal works wonders. Thanks for posting this issue for me to find it as I was losing my mind.

krisdover, 2022-07-26:
@fenix8k thanks so much for this, I was starting to think I got a board with a broken touch screen. 😅
A bit of tape around the edge of the metal enclosure and now it works fine.