bigtreetech/BTT-TFT35-E3-V3.0 #14: NEW - TFT35 E3 V3.0 with SKR Mini E3 V1.2

midmadn, 2020-04-20:
I purchased this combo from Amazon. The Skr Mini works well as confirmed by connecting my stock Ender 3 Pro touchscreen.
But the TFT35 E3 V3.0 just sits there waiting for me to select TouchScreen or LCD mode. The knob seems fine but when I turn it nothing happens. When I press it nothing happens unless I hold it for a few seconds. Then I get a steady tone. I tried updating the firmware with the SD Card and it updated the firmware and all of the icons to switch to material theme. But still ends up waiting for me to select touchscreen or LCD mode. No input of the knob changes it.

n3rdz, 2020-04-20:
same here, purchased 20.04.2020 on amazon

midmadn, 2020-04-21:
Working great now.

closed 2020-04-21 by midmadn