bigtreetech/BTT-TFT35-E3-V3.0 #16: Babystepping not available in TFT mode

JoeK1973, 2020-05-09:
I am on the latest firmware on both TFT35 E3 v3 (3.26.1) and Marlin 2.0.5.3 on SKR Mini E3 v1.2

I have babystepping available on Marlin mode but not in TFT35 mode.

oldman4U, 2020-07-03:
Hi.

Could you please test with the latest "Master" firmware of the TFT. I can execute Babystepping while printing and change it from TFT during a print.

oldman4U, 2020-07-11:
Hi. 

Could you please reply or close this ticket in case the issue is solved in the meantime. 

Thank you

JoeK1973, 2020-07-11:
Hi,

Yes that fixed it.

Thanks

oldman4U, 2020-07-11:
👍Great. Please close the ticket 

Thank you

oldman4U, 2020-07-18:
@bigtreetech This issue is solved but the user is not able to close it for an unknown reason. Could you please close it. 

Thank you

oldman4U, 2020-08-05:
With the latest release it is also possible to save the babysteps. 

Please close this ticket

closed 2020-09-23 by JoeK1973