bigtreetech/BTT-TFT35-E3-V3.0 #17: Weel steps.

Heartlander81, 2020-05-24:
What must i write in Marlin for the weel steps at the encoder? I mean "encoder pulses per step" and "encoder steps per menu item"
Must i reverse the encoder direction, menu direction ore Both option?

Thanks.

oldman4U, 2020-05-29:
Hi.

I never changed such a setting and all my encoders work fine. What is the problem or what would you like to achieve?

Heartlander81, 2020-05-29:
Hi. I have changed my Marlin Firmware and now is all good.

I think the problem ist in my config. Then i have the original tvt from
anycubic and the tft 35. I want run both tft at same time. But under Marlin
2.0 is the tft 35 my favorit and i have disconnected my anycubic tft.
Thanks for your requests.
I wish u a nice time.

Sven Baum

oldman4U, 2020-05-31:
Sounds great.

Could you please close this ticket in case you think it is not needed any more.

Thank you

closed 2020-05-31 by Heartlander81