bigtreetech/BTT-TFT35-E3-V3.0 #18: No connection diagram for tft35 e3

KKrossbow84, 2020-06-16:
No diagram to assist connecting TFT 35 E3 to stock creality motherboard

oldman4U, 2020-06-16:
Hi.

As far as i know, you disconnect the single cable from to the original LCD display ( i assume this is the display you use at the moment) and connect it to EXP3 of the TFT.

I expect you are aware, that this only gives you Marlin mode and a mainboard with serial connection is mandatory for touch screen functionality.

Hope this help

oldman4U, 2020-06-19:
Hi.

Any news?

thirdender, 2020-11-27:
Would it be possible to document that a serial connection is necessary for the touch screen in the README? It might be helpful to list common troubleshooting steps and pitfalls for future visitors. I'd be willing to submit a PR if you like.

oldman4U, 2020-11-27:
Hi. When have you read it the last time. There was an update which included this at the beginning of the documentation:

Do the following, to be able to use the touchmode of your screen.

Connect the 5pin serial cable according to the manual of your mainboard.
Define a serial port in Marlin, to activate the port used by the TFT.
Make sure the same BAUDRATE is defined in Marlin and the firmware of your TFT (config.ini)

thirdender, 2020-11-27:
I have the 4.2.2 version of the stock Ender 3 Pro motherboard. I'm having trouble finding a diagram for that board. As you pointed out, not all mainboards have a serial connection. Do you know if the v4.2.2 stock Ender 3 Pro board has a serial connection?

oldman4U, 2020-11-28:
Sorry but I don't know.

siral1, 2020-12-11:
I also have the 4.2.2 version of the stock Ender 3 Pro motherboard. TFT35 v3 E3 does not work. Does anyone have advice?

radek8, 2020-12-25:
Original Ender boards have limitations.
I recommend replacing the original board with a suitable alternative
for example:
SKR Mini E3 V1.2
SKR mini E3 v2.0
SKR E3 DIP V1.1

siral1, 2020-12-25:
#ssc45289 #response_container_BBPPID{font-family: initial;font-size: initial;color: initial;}  I already know, I ordered a SKR 1.4 turbo with TMC 2208. Thank you very much.                                                                                                                                             Odesláno ze smartphonu BlackBerry – nejzabezpečenějšího mobilního zařízení              Původní zpráva       Zobrazit podrobnosti                                                                                                                                                                         Od:  notifications@github.comOdesláno: 25. prosince 2020 22:39Komu:  BTT-TFT35-E3-V3.0@noreply.github.comOdpověď komu:  reply@reply.github.comKopie:  laris.rene@gmail.com; comment@noreply.github.comPředmět: Re: [bigtreetech/BTT-TFT35-E3-V3.0] No connection diagram for tft35 e3 (#18)  
Original Ender boards have limitations.
I recommend replacing the original board with a suitable alternative
for example:
SKR Mini E3 V1.2
SKR mini E3 v2.0
SKR E3 DIP V1.1

radek8, 2020-12-26:
Good choice :-)
But you will have to make a new cover for the electronics.
I have it too, but I replaced the TMC2208 with the TMC2209.

oldman4U, 2020-12-26:
Thought you do not use sensorless homing!??

radek8, 2020-12-26:
I have no use sensorless homing.
I tested it, but the sensitivity changed after each marlin update. and calibration was required.
TMC 2209 with less heating.

oldman4U, 2020-12-26:
Ok. But nobody prints as fast as you print. 🤪