bigtreetech/BTT-TFT35-E3-V3.0 #19: Disabling steppers after printing from tft

mine2209, 2020-06-29:
I cannot stop the tft disabling the stepper motors after every print which causes print head to drop into work piece. 

I have checked the stepper timeout, sd print release settings, and end gcode. No m18s or m84s present in start/ end gcode or the print gcode file. If I print from PC via cable & pronterface I do not get this issue using same gcode files so believe it must be tft causing the issue. It also happens after usb print via tft.

mine2209, 2020-06-29:
There is in fact a M84 S0 in gcode file but this I believe is disabling timeout and I have tried removing it and it did not solve the issue.

oldman4U, 2020-07-01:
Hi.

By default there is a M18 in the end gcode. Sure you removed it? Which version of the firmware are you using? Maybe you have to change this in config.ini!

mine2209, 2020-07-01:
No M18 in there I did check here is the gcode 

`; short gcode for test
; Safe Z height: 6.000
; Tools: 1 = V-Bit (90 deg 0.5")
; Notes: 
; Generated Monday June 29 2020 09:06 PM
 
G90
M03 S16000
G21
G00 X0.000 Y0.000 Z0.000
G1 Z6.000 F500
G1 X0.000 Y0.000 F2540.0
 
; Tool 1: V-Bit (90 deg 0.5")
; Path: Profile 1 F:\short gcode for test.gcode
; 
G0 X443.500 Y-17.001 Z6.000 F5080.0
G1   Z-1.000 F762.0
G1 X498.653 Y44.150  F2540.0
G1 X443.500 Y105.301  F2540.0
G0   Z6.000 F5080.0
G1 6.000 F500 ;goto safe z`

made above as test file and still does it.

Checked config.ini which is adjusted here is the current config file im using have a look but i cant see what is causing the issue in there.

`#### Default Config file for BigTreeTech TFT Controllers
#
# BIGTREE_TFT35_V1_1 / BIGTREE_TFT35_V1_2    / BIGTREE_TFT35_V2_0
# BIGTREE_TFT35_V3_0 / BIGTREE_TFT35_E3_V3_0 / BIGTREE_TFT28_V1_0
# BIGTREE_TFT28_V3_0 / BIGTREE_TFT24_V1_1    / MKS_32_V1_4
#
# Firmware Source: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

#--------------------------------------------------------------------
# Supported Marlin Firmware Version
#--------------------------------------------------------------------
#  Minimum Marlin Firmware Version: Bugfix-2.0.x
#  Distribution date: 2020-05-12
#  Source: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

#--------------------------------------------------------------------
# General Settings
#--------------------------------------------------------------------

#### UNIFIED MENU / CLASSIC MENU
# Select a UI Menu flavour
# Options: [Unified Menu: 1, Classic Menu: 0]
unified_menu:1

#### Baudrate
# Options: [2400: 0, 9600: 1, 19200: 2, 38400: 3, 57600: 4, 115200: 5, 250000: 6, 500000: 7, 1000000: 8]
baudrate:6

#### Default Touch Mode Language
# Select the language to display on the LCD while in Touch Mode.
# Options: [0: ENGLISH,   1: CHINESE,  2: RUSSIAN,  3: JAPANESE,  4: ARMENIAN,
#           5: GERMAN,    6: CZECH,    7: SPAIN,    8: FRENCH,    9: PORTUGUESE,
#          10: ITALIAN,  11: POLISH,  12: SLOVAK,  13: DUTCH,    14: HUNGARIAN]
language:0

#### Default Touch Mode Color Options
# Options: [0: WHITE,   1: BLACK,    2: RED,    3: GREEN,   4: BLUE:,      5: CYAN,        6: MAGENTA,  7: YELLOW,
#           8: ORANGE,  9: PURPLE,  10: LIME,  11: BROWN,  12: DARKBLUE,  13: DARKGREEN,  14: GRAY,    15: DARKGRAY]

# Title background color
title_back_color:1
[... 309 lines cut ...]
# Options: [enable: 1, disable: 0]
start_gcode_enabled:0

#### Default End Gcode Status
# Options: [enable: 1, disable: 0]
end_gcode_enabled:1

#### Default Cancel Gcode Status
# Options: [enable: 1, disable: 0]
cancel_gcode_enabled:1

#### Start G-code - run this G-code before starting print
# maximum length 50 characters
start_gcode:G28 XY R20\n

#### End G-code - run this G-code after finishing print
# maximum length 50 characters
end_gcode:G1 X0 Y0 Z50\n

#### Cancel G-code - run this G-code after canceling print
# maximum length 50 characters
cancel_gcode:M0\G1 Z10\nG1 X0 y0\n
`

Thanks for suggestion though.

mine2209, 2020-07-01:
Believe the firmware is most up to date 0.26 30 may 2020? 

i have also checked my marlin firmware and sd disable is definitely off here is the code from firmware i have flashed 

`  // around this by connecting a push button or single throw switch to the pin defined
  // as SD_DETECT_PIN in your board's pins definitions.
  // This setting should be disabled unless you are using a push button, pulling the pin to ground.
  // Note: This is always disabled for ULTIPANEL (except ELB_FULL_GRAPHIC_CONTROLLER).
  #define SD_DETECT_INVERTED

  **#define SD_FINISHED_STEPPERRELEASE false          // Disable steppers when SD Print is finished
  #define SD_FINISHED_RELEASECOMMAND "M84 S0" /**/ You might want to keep the Z enabled so your bed stays in place.

  // Reverse SD sort to show "more recent" files first, according to the card's FAT.
  // Since the FAT gets out of order with usage, SDCARD_SORT_ALPHA is recommended.
  #define SDCARD_RATHERRECENTFIRST

  #define SD_MENU_CONFIRM_START             // Confirm the selected SD file before printing

  //#define MENU_ADDAUTOSTART               // Add a menu option to run auto#.g files

  //#define EVENT_GCODE_SD_STOP "G28XY"       // G-code to run on Stop Print (e.g., "G28XY" or "G27")

  /**`

as you can see above sd disable is off and if it was on the gcode it runs is to disable the timeout...

oldman4U, 2020-07-02:
I like the motors to be deactivated after print, so i never cared. But i hope i can do some tests soon to check if the End gCode makes a difference. Somehow it has to, otherwise the function would be useless.;-(

mine2209, 2020-07-02:
End gcode does work, It runs the end gcode I program and then after finishing the end gcode on the tft it disables the steppers.

My gantry is heavy so It drops into the work. I have to catch it before it damages the workpiece. 

I need it to remain in place so I can do toolchabges without having to repetitive position which invariably adds error

oldman4U, 2020-07-02:
Hi.
Maybe a workaround for now.

I replaced the whole End gcode in config.ini with a simple M17 and now ONLY the motor of the Y-axis stays activated.

Let me know

oldman4U, 2020-07-05:
Hi.

Any news regarding the reported issue?

Please let us know

mine2209, 2020-07-05:
Hi, sorry for the late reply. I saw your initial message before edit regarding Y which wasnt going to help so I didnt try. Just seen your edit now to Z 

interesting progress.

Just M17 in end gcode in config.ini without \n does prevent the issue, left all the steppers active for me not just Z. I then will just need to manual move the head to next location.

The interesting part now...

If I add M17\n the issue remains

So i tried adding my toolchange location with out the \n and the steppers remain active and it doesn't move to the location so I can only assume that when I put just M17 that it isnt the M17 keeping them active but the lack of the \n which must enter the code, I assume it isnt even running the M17. 

Issue isnt solved but this means at least I can get on with my project

mine2209, 2020-07-05:
Cancel that it doesn’t disable the steppers but it locks me out of the machine. When I try to manually move the carriage I get busy processing, please wait at the top

oldman4U, 2020-07-05:
Not sure there is a logical explanation for all this😜

So this locks you out generally or once the print has finished?

What happens with the stepper motors when you cancel  a print?

And have you mentioned if you print from mainboard or TFT?

guruathwal, 2020-07-06:
The `M18` g-code is hard-coded in the firmware in the printing.c file. You will have to change it in the firmware and recompile it again for now. 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/868d61549cc480a06c6c2526b828efab5807539a/TFT/src/User/API/Printing.c#L96-L110

oldman4U, 2020-07-06:
Hi Gurmeet.

I knew you could tell us, and I even checked Print.c but did not know there is a Printing.c...;-( Every day something new to learn.

Wouldn't it be good to mention the commands which make sense to use as Start/Pause/End gcode. I mean, it makes no sense to be able to manipulate a function which is changed by a setting which is much deeper in the system right after that.

mine2209, 2020-07-06:
Thanks, I was using pre complied firmware, I will have a look at modifying it and recompiling, hopefully just as simple as changing the text or I will be over my head lol only normally change configuration options not base code.

Why give the option for end gcode if the firmware is going to override it. As a suggestion a better option maybe to place these "idle commands" in the endgcode section and just have it enabled by default that way it is easily edited by enduser and visible should people wish to stop it.

Thanks for the help, I will let you know how I get on when I get chance to look at it.

oldman4U, 2020-07-07:
Unfortunately this needs more than changing the config.ini. But not to compile the own software means skipping the funniest part. I would highly recommend to give it a try.

oldman4U, 2020-07-09:
Hi.

Today I found some time to test the fix Gurmeet recommended, and it works perfectly. If it is ok for you, I will start a ticket, so this gets fixed one day.

As far as I know, I am not allowed to offer you a fix here, by sending you a compiled version. But if you send me your email address, I can do so.

Please let me know

Thank you

oldman4U, 2020-07-11:
Hi. Any news regarding the reported issue

oldman4U, 2020-07-15:
Hello mine2209.

Could you please close this ticket or answer my question above.

Thank you

mine2209, 2020-07-18:
Sorry for the lack of replies I have been personally affected by the global pandemic recently and have not been able to test anything. I will close this for you now and as soon as I work out how to send you my email I will do as that would be a great help.

thanks for your help again.

closed 2020-07-18 by mine2209