bigtreetech/BTT-TFT35-E3-V3.0 #20: TFT35 E3 on SKR mini E3 -- Touch screen mode not available

kschoone, 2020-07-08:
Hi,

I have a SKR mini e3 v1.2 and now received the TFT35 E3.
After connecting the TFT to the Mini E3 I do get the Marlin screen version just fine.
But when I press the selector button it shows both modes, but I am not able to select the Touch screen version.
If I only connect the TFT ribbon I do not get any screen on the display.

In Marlin mode everything works fine.

Can you please help me?

oldman4U, 2020-07-08:
Hi. 

As far as i know the skr e3 has no serial port and therefore you can not use the touch mode. You need a mainboard which supports the touch mode.

oldman4U, 2020-07-08:
Like the SKR E3 DIP.

kschoone, 2020-07-08:
The skr mini e3 has the port for IT.
I have it working now. I updates the firmware for the tft35 to do that

oldman4U, 2020-07-08:
You are right. It is besides the Servos. When you update the firmware do not forget to update the font and the bmp and the config.ini also.

Please close this ticket once everything works properly

closed 2020-07-08 by kschoone