bigtreetech/BTT-TFT35-E3-V3.0 #22: freze when selecting print from usb or sdcard

mithae, 2020-07-20:
I have a skr mini e3 1.2 and a tft 35 e3 v3.0 installed in my ender 3pro. When i try to select a print job in touch mode. The screen screen freezes when it ask me if iam sure i want to print this. I cant figure out why? i have updated firmwaware several times. But the problem remains.

oldman4U, 2020-07-21:
Hi mithae

Which firmware have you updated and to which version. I use the same TFT with a SKR E3 DIP mainboard and it works great.

oldman4U, 2020-07-21:
And tell me more about the SD card? Which size and format?

mithae, 2020-07-21:
Hi!
I have tried all the sd cards i have.  So One card is kingston 16gb fat32. One came with the skr board and its no name micro 128mb and just fat. The firmware version on the tft is "BIGTREE TFT35 V3.0 E3.26" Or thats the name of the file ive used here on githube.  I can print from marlin mode with no problems at all.

mithae, 2020-07-23:
Ive figure out that if the file name is too long, then the screen will freeze. Is there something i can change in the config file to make it better... Or should i just rename it in cura?

oldman4U, 2020-07-23:
How long is too long?

oldman4U, 2020-07-26:
This seems to be too long😁

Please close this ticket 

Thank you

mithae, 2020-07-26:
Sorry for not answering... Dont know how many letter in the name that is max. But now i change them in cura... So now it works very well... :) Thanks for the support!

closed 2020-07-26 by mithae