bigtreetech/BTT-TFT35-E3-V3.0 #23: touch screen does not light up or work

wkrogers, 2020-07-21:
I have an ender 3 pro.  I just upgraded the controller board to an SKR Mini E3 V2.0+TFT35 E3 V3.0.  When powering up can hear fan but no display.  Hook up original display and it works perfect.  Any suggestions

oldman4U, 2020-07-22:
Hi. How have you connected the display. Maybe a picture could help.

oldman4U, 2020-07-24:
Any news? Please let us know. Thank you

oldman4U, 2020-07-26:
Could you please close this ticket in case you found the problem. 

Thank you

oldman4U, 2020-07-29:
@bigtreetech 

Could you please close this ticket. Seems this user does not know how to do this.

Thank you

oldman4U, 2020-09-23:
Please help the community and close the ticket once you do not need it anymore. Thank you

wkrogers, 2020-10-04:
how do i close?  Found cable that came with display bad.

oldman4U, 2020-10-04:
Using a web browser you have a Close button at the bottom of the ticket. On iOS you can drag the line right above the Comment up, wich opens a window like shown at attached screenshot. [image]

Thank you

wkrogers, 2020-10-06:
That is on a celll phone - I don't use a cell phone

oldman4U, 2020-10-07:
In the browser on PC you click Close or Comment and Close.

closed 2020-10-13 by wkrogers