bigtreetech/BTT-TFT35-E3-V3.0 #25: TFT unresponsive when mounted to chassis

ventoetc, 2020-08-04:
The screen works fine until you screw it into the screen display using the 4 M2.5 screws. This is for and Ender 3 pro and this is the 2nd copy of the LCD that behaves identically. 

When you unmount the display from the frame, it works just fine.

oldman4U, 2020-08-05:
Hi. This is a known issue which can be solved easily by insulating the tft and the metal frame using a bit of insulation tape. Could you please make a picture of the back of the tft how you got it from btt and upload it here. 

Thank you

ventoetc, 2020-08-05:
[image]

That makes sense, but i don't see where it is touching the metal? It looks like the board only makes contact with the standoffs.

I will insulate around the outside of the board and remount.

oldman4U, 2020-08-06:
Insulate also where the standoffs Touch the display with tape or a plastic
washer.

oldman4U, 2020-08-09:
Hi. Hope everything works properly. Please do not forget to close this ticket 

Thank you

ventoetc, 2020-08-13:
After insulating the front of the panel, the shorting issue has stopped.

closed 2020-08-13 by ventoetc