bigtreetech/BTT-TFT35-E3-V3.0 #26: TFT35 Blank Screen and non Responsive

pedrax202, 2020-08-08:
Very similar to another ticket here I have recently added an SKR mini E3 v2.0 board and TFT35 V3 to my Ender 3 Pro. Both were working perfectly for about 2 or 3 days then during the middle of a print the print stopped and the TFT when blank. Rebooting didn't help, I tried to reflash firmware and it wouldn't take.
Power LED was lit on back of board and it would recognise an SD Card (green LED). The printer would fail to initialise as Octoprint could not see it. As soon as I disconnected board and installed the original Creality screen the printer boots fine and operates normally.

TFT was connected by EXP3 and RS232 to the E3 V2 board by TFT and EXP connection. Could not reset TFT and no other attempts to restore have worked.

oldman4U, 2020-08-08:
Have you tried to remove the TFT from the metal plate it is mounted on?

pedrax202, 2020-08-08:
Yes. Tried with it removed aswell. No improvement

oldman4U, 2020-08-09:
Could u send a picture which shows the screen when it is white. Have you tried with only the serial connected  Not sure it works with only the exp connected but i can try.

oldman4U, 2020-08-09:
OK, so I tested my screen with only the serial and only the EXP connected and in both cases the display turns on and shows the user interface (touchscreen).

Seems to be a broken mainboard, cable or TFT...;-(

Please let me know

pedrax202, 2020-08-09:
> 
> 
> OK, so I tested my screen with only the serial and only the EXP connected and in both cases the display turns on and shows the user interface (touchscreen).
> 
> Seems to be a broken mainboard, cable or TFT...;-(
> 
> Please let me know

Hi Oldman4U. Tried each of the combinations and my screen stays off. Doesnt even illuminate. With the 10 pin cable in EXP3 the Power LED comes on but nothing happens. The printer will also not boot when the TFT is connected. I have contacted BTT through Aliexpress to try and get a replacement.

oldman4U, 2020-08-09:
Makes sense. Please close this ticket as the problem seems to be hardware
related.

Thank you

closed 2020-08-09 by pedrax202

AlexKBR2, 2023-07-23:
Hey guys, I was so disheartened by this crappy problem, that I connected a wire to the 5v of one of the exp slots on the back of the tft35 and it connected randomly to the other pins, and, it came back to life! But it only works if I turn on EXP1 and EXP3... I don't know why. In this case, I used my old RAMPS 1.4. Even so, it is not behaving normally... The Display resets itself, and in the beginning the functions were all messed up, with the icons not corresponding to the functions, then I updated the firmware again and they returned to their normal positions. The two images below...
[image]
[image]