bigtreetech/BTT-TFT35-E3-V3.0 #28: No Printer Attached

Ripper80, 2020-08-18:
I have the TFT35 E3 v3.0 attached to the SKR 1.4 Turbo w/TMC2209 drivers. I have followed directions from several YouTube videos, like Teaching Tech, and also posted in several locations the issue that I have with this combination.  People keep telling me it is operator error, but then that is why I have been asking for help everywhere.

I have RS232 connected to TFT, have both the EXP1 and EXP2 connected to same ports on the SKR and TFT.  Get both the Marlin and Touchscreen, but the TS reports NO PRINTER ATTACHED.  I have tried BAUDRATE of 115200 and 250000 and ensured that they are the same on both SKR and TFT.  Folks recommended that I invert the cables attached to the EXP ports....did that, the screen is blank.  Folks recommended that I invert the RS232 on the SKR, did that, but then the screen is either flashing or blank depending on where I put the reset cable.  Folks recommended disconnect the reset cable on RS232, did that and the screen remains blank.  Inverted the cable and RST disconnected, screen blank.  I have tried the CR-10 connection from EXP1 to EXP3, no display.

In Marlin, I have left PORT at 0 and enabled the PORT_2 to -1, I have changed PORT to 1 and PORT_2 to -1, PORT to 1 and PORT_2 to 0, PORT to 1 and PORT_2 to 2.

Folks have been quick to indicate that their system works without issue....but then they are using TFT35 v2.0, have the SKR mini or some other combination, which is all nice and good....but those of us with the combination I indicated when I started this post continue to get NO PRINTER ATTACHED.  If you have this combination of equipment, I would like to see pics of your cards and if you don't mind sharing, a copy of the firmware source.  I KNOW I am doing something wrong, but in my limited knowledge of the boards and the firmware mods, it is something trivial, but beyond my knowledge to find.  THANK YOU for any help you can provide.

Roughd1433, 2020-08-18:
I am using the same set-up as you. My serial port settings are port_1 is -1
and port_2 is 0.
I have my baud rate set at 250000.

I made sure I had matched up the 5v pins between the rs socket and the tft
just in case the wires in the lead were wrong.

Also when you change the baudrate on the screen don't forget to hit the
back button till your on the home screen then wait a few seconds for it to
see the printer.
Hope this helps

Ripper80, 2020-08-20:
Tried the port settings you have, same message; NO PRINTER ATTACHED.  Let the printer sit in that mode 30 minutes, no change.

thenewdays, 2020-08-24:
I have the same issue, what I found that worked was to remove the exp cables and turn off the printer and back on, no power to screen or board, turn on then adjust com speed to 115200 or 250000 and back 1 screen, wait 5 seconds. If it doesn't remove the no printer attached the do again. For me it finally works. If I don't leave the printer on I have to do it all again. Using marlin 2.0.6 and tft v3.

oldman4U, 2020-08-27:
Hi Ripper80.

The display has to work without such tricks.

If it does not, usually this is related to a wrong setting in Marlin or the TFT firmware (wrong display type, wrong baud rate, wrong serial port, mismatch of config.ini and firmware, and so on) or it is hardware related (bad cable, broken mainboard, broken TFT).

We have also seen many issues where the TFT touched the metal frame of the Ender, removing the TFT from the frame should help in this case, and we also saw issues where the power from the power supply was not efficient enough or broken.

Hope this helps

oldman4U, 2020-09-07:
@Ripper80 

Please close this ticket in case the problem is solved in the meantime.

Thank you

MaasonR, 2020-09-07:
Whwn there is a solution to this problem, I will be happy to close it... To date, none of the recommendations has worked and the issue still exists.  I have tried multiple things in Marlin without success. Purchased yet another TFT that hasn't been mounted and has been isolated so that it does not accidentally touch the metal of the printer, have swapped cables, inverted cables, replaced cables and changed the bad rate to several different rates, all without success.

That the RS232 cable is so convoluted and confusing (all black with cryptic markings that are not defines) and other things that may be the cause of the issue; I want a solid fix, not speculation from users.  Why can't the vendor provide a solid solution to this issue which is not isolated to me?  And you claim you are new to all this... Why are YOU pushing me to close the issue?

oldman4U, 2020-09-07:
@MaasonR 

Are you Ripper80? If not, you can not close this ticket and only Ripper80 (and BTT) can do this.

No-one is asking anyone to close an active ticket, where the problem could be software related. But i ask Ripper80 to close it, because he does not respond to his own ticket, which is not very friendly. In case he was able to solve the problem, he could let us know and close it.

The single strings of the RS232 cable could be completely without any marks. It is just important to connect the corresponding parts of the connector on the mainboard and the TFT correctly. I used current wires all with the same color to make my own long cable.

The hardware is made by BTT and the software is made by the open source community using this GitHub. I am not working for BTT and also i do not get paid by then or someone else. I am a user like you who understands how OpenSource works and that it is important to give something back - if possible. That i am new to 3D printing does not mean that i can to try to help.

Ripper80, 2020-09-07:
Not sure why I have two profiles...one apparently on my smartphone and another on my desktop...Will correct that.

Since the issue has not been determined hardware or software, leaving it open.

oldman4U, 2020-09-08:
Sure.

Last view days similar issues have been solved because the cable where wrong and because cr10 and reprap display where activate in firmware.

Maybe it is wort to check this again.

oldman4U, 2020-09-12:
So is it your goal to find the problem and to solve it, or to keep the ticket opened?

Ripper80, 2020-09-13:
Tried your port settings, but they didn't help.  two new screens and two new motherboards, the issue remains the same.  After manipulating the RS232 cable, I managed to get the NO PRINTER ATTACHED, but the screen flickered and I couldn't do anything on the touchscreen. I couldn't control anything. Of course, I can get a crystal clear screen, but since NO PRINTER ATTACHED, can't control the printer then either...but it is pretty.

Since this topic is for BTT, I would assume that at some point in time, someone from BTT will actually see this problem and see if they can work with me to resolve the issue.  Am I missing the whole reason for this ISSUE category for the BTT components?  Only one I consistently get responses from has NO CLUE how to fix the issue and it appears their only goal is to close open issues, whether the problem is solved or not...

The issue will be closed when someone else closes it or I get/find a solution to the problem...NOT UNTIL THEN.  I don't care if it takes MONTHS...when you pay good money, you should get help when the products don't work.

radek8, 2020-09-20:
Unfortunately, I'm not from BTT
I have a question.
You do not write anywhere if the Marli screen is available when you switch to the Marlin view and connect the EXP1 and EXP2 cables.

Zed7Seven, 2020-10-31:
Ender 3 V2 Printer not attached! I had the same issue when I installed SKR mini E3 V2, with The V3 Display. On the BTT screen, Ensure Baudrate is at 115200. Then check either your cables connecting to the thermistor (two white cables/last connection on the far right corner of your mainboard (SKR mini V2). This could be a thermal error that has to do either with your thermistor probe or cartridge connected to your hot end. A replacement may be needed. I hope this helps!

Ripper80, 2020-11-02:
NON RESPONSE from vendor and the responses from othrs are of no help as they are using different equipment and setups....READING IS FUNDAMENTAL.

closed 2020-11-02 by Ripper80

radek8, 2020-11-02:
Don't expect help from Bigtreetech. Bigtreetech produces HW users compile SW :-)
You are asking for help and you will not answer questions.
You also did not upload your configuration as required when creating the ticket.
I have similar HW as you, but I have the impression that you don't need help, you just wanted to complain.

radek8, 2020-11-02:
In that case, good luck in seeking advice on Youtub

eguvenn, 2021-02-20:
Hello I have the same issue. I believe this is a firmware issue.  While I have the  screen reporting  no printer attached,  I connect through PC over USB,  with Repetier SW.  Then  screen turns into  ready.  However there is no functionality.

Roughd1433, 2021-02-22:
> Hello I have the same issue. I believe this is a firmware issue. While I have the screen reporting no printer attached, I connect through PC over USB, with Repetier SW. Then screen turns into ready. However there is no functionality.

try swapping the exp cables over. and also set the baud rate on the connection tab to 115200.

radek8, 2021-02-22:
@eguvenn 
Create your own ticket, write what kind of HW you have, describe the fault in detail, attach configuration files and you can attach photos of how the cables are connected.
Thank you

xynyxs, 2021-02-25:
I see lots of posts on different places on this isue, both in exactly this configuration and slighty different combinations of mainboard and screens. In my case my new Mingda D2 (with SKR v1.3 mainboard) gave this message (plus "unknown command" in case of trying microsteps) from the beginning. It dissapeared when moving one off the axes, but after two days it remained for ever, leaving me with a useless factory configured printer... Please stop bashing people wrestling with this frustrating problem

radek8, 2021-02-25:
> I see lots of posts on different places on this isue, both in exactly this configuration and slighty different combinations of mainboard and screens. In my case my new Mingda D2 (with SKR v1.3 [...]

For most users, the display works without major problems. Those who can do this are trying to eliminate the reported problems. Unfortunately, most of the problems you experience are caused by poor wiring, poor configuration,
incompatible Marlin or TFT settings or defective HW.
For example, your unknown command error means that you are using a command that Marlin does not have enabled. If you have a problem that you want to solve, what I wrote in the previous post also applies to you.

xynyxs, 2021-02-25:
I would agree with your analysis if I hadn't been confronted with it in a factory build printer where I must assume that not every user is affected although they have the same firmware. And of course I checked the (hot glued) wiring (leaving it untouched for guaranty reasons). Hope they come up with a solution ...

joemama626, 2021-02-28:
I have this exact same set up with the same issue.  Can't find a solution either.  Can anyone help?

sogeniusio, 2021-03-06:
Not sure if this is the same for TFT35 but I got it to work on my TFT70 - Marlin-bugfix-2.0.x

```
Set baud to 250000
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define SERIAL_PORT_3 6
```

After setting those, and switching baud on the tft, it came right up!

Hope this helps!

radek8, 2021-03-06:
> Not sure if this is the same for TFT35 but I got it to work on my TFT70 - Marlin-bugfix-2.0.x
> 
> ```
> Set baud to 250000
> #define SERIAL_PORT -1
> #define SERIAL_PORT_2 1
> ```
> 
> After setting those, and switching baud on the tft, it came right up!
> 
> Hope this helps!

The display does not matter.
The serial port number depends on the Marlin board used.
Different manufacturers use different port numbers.
It can be found in the documentation or on the forums.
So if he has the same board as you, it will help him, unfortunately you do not mention it as a board with Marlin you have ...

sogeniusio, 2021-03-06:
> The display does not matter.
> The serial port number depends on the Marlin board used.
> Different manufacturers use different port numbers.
> It can be found in the documentation or on the forums.
> So if [...]

Ahh I just updated my comment. I also found this information here from one of the FB groups: 

https://www.dropbox.com/s/k7tb4sck5d2h5bb/BTT%20TFT%20to%20BTT%20Mainboards%202021.pdf?dl=0

radek8, 2021-03-06:
useful table  :+1:

radek8, 2021-03-07:
> Ahh I just updated my comment. I also found this information here from one of the FB groups:
>
> https://www.dropbox.com/s/k7tb4sck5d2h5bb/BTT%20TFT%20to%20BTT%20Mainboards%202021.pdf?dl=0

@sogeniusio do you have a contact for the person who created the PDF file you linked to?
If the author gave permission to publish in this repository, it could help more users.
Thank you

radek8, 2021-03-07:
> Hello I have the same issue. I believe this is a firmware issue. While I have the screen reporting no printer attached, I connect through PC over USB, with Repetier SW. Then screen turns into ready. However there is no functionality.

@eguvenn
look here, you have the same display behavior
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

radek8, 2021-03-07:
> I have this exact same set up with the same issue. Can't find a solution either. Can anyone help?

@joemama626
Open your own ticket, describe the problem, write what configuration you have, publish the configuration files. You can add photos of the connection
I have the same problem with the shout, it is not reasonable to answer.
Thank you for understanding

qu4droro, 2021-09-10:
> Not sure if this is the same for TFT35 but I got it to work on my TFT70 - Marlin-bugfix-2.0.x
> 
> ```
> Set baud to 250000
> #define SERIAL_PORT -1
> #define SERIAL_PORT_2 1
> ```
> 
> After setting those, and switching baud on the tft, it came right up!
> 
> Hope this helps!
> 
> Edit: I'm using an SKR Pro 1.2

Im using SKR Pro and it worked for me!! Thanks for helps...

snke95, 2022-02-27:
Salve, io ho avuto lo stesso problema su mks gen l2.1 , ho poi inverito i cavi rx e tx del cavo aux1 e  ah subito riconosciuto la stampante

su marlin le impostazioni sono
#define SERIAL_PORT 0
#define BAUDRATE 115200

gibbermagash, 2022-05-18:
Based on this video [Big Tree Tech SKR 2 - Install - Marlin Configuration - Chris's Basement](https://www.youtube.com/watch?v=1PPBHn-jU0k)  Check the inspection sticker on the back of the board.  If there is just one green sticker it is likely one of the boards that has a bad mosfat chip.  If there is a green inspection sticker and chrome inspection sticker it's probably still good.