bigtreetech/BTT-TFT35-E3-V3.0 #30: Display not functional

petenut, 2020-08-23:
Hello, I am upgrading my display from V2.0 to V3.0. The V3.0 display LED backlights turn on, but the display does not show any image and does not seem functional. Please help.

oldman4U, 2020-08-23:
Which mainboard?
Pictures of the cables?
Firmware version?
What else is attached to mainboard and TFT?

petenut, 2020-08-24:
Hello,
   Sorry for not including the details.

AnyCubic Kossel Linear

BTT SKR 1.3 main board
 - heated bed
 - hot end
 - driver cooling fan
 - Marlin 2.0

No other peripherals connected.

I have tried only using the serial cable, only the EXP cables and a combination of both sets of cables.

The backlight turns on,  but the display doesn’t function.

Your help is much appreciated.

Thank you,

J Petrie

oldman4U, 2020-08-24:
Looks like a hardware issue to me. Have you tried to upload a new firmware?

petenut, 2020-08-25:
Hello,
   I just tried to flash the firmware but same result. Backlight is on but the screen is black. Can I exchange this unit?

Thank you,

J Petrie

oldman4U, 2020-08-25:
Can you see the firmware flashing?

Contact your dealer for exchange.

petenut, 2020-08-27:
Hello, I have returned the original display item and have received a new one. I haven’t had a chance to test the new display.

J Petrie

oldman4U, 2020-08-28:
Let me know if it is fixed.

petenut, 2020-08-29:
Hello,
    I have received the replacement display and I’m able to get the graphics menus to work. When I push the encoder knob I get the option to switch to Marlin but the screen goes black when I select it. I’m able to hold the knob down and go back tot the graphical menus but I’m not happy because I bought this display specifically to access the Marlin options. I’ve read one of the guides on Github but I don’t know what the solution is to fix this problem.

Please advise.

Thank you,

J Petrie

oldman4U, 2020-08-30:
Have you tried to rotate the knob once you entered Marlin mode? This is needed to switch from black to screen. Let me know.

petenut, 2020-08-30:
Hello, yes I tried to rotate the knob and press the knob after changing modes but the Marlin menus still don’t appear. What should I try next?

Thank you,

J Petrie

oldman4U, 2020-08-30:
How have you connected the TFT with the mainboard?

petenut, 2020-08-30:
Yes sir. I have all three cables (Exp 1, Exp 2 and TFT) connected to the main board.

J Petrie

oldman4U, 2020-09-03:
Sorry for the late answer.

Do you compile firmware by yourself or do you use precompiled firmware?

petenut, 2020-09-04:
I’ve actually been having a hard time getting the firmware to compile so I’m stuck at the moment.

What software do you recommend? I have VS code/PIO.

Thank you,

J Petrie

oldman4U, 2020-09-04:
I use the same on Mac.  What is the problem with the comp. ?

oldman4U, 2020-09-07:
When you have a problem with compiling, try to delete the platforms and packages folder. On the Mac this can be found under: /Users/username/.platformio/ - which is a hidden folder.

Start VSC after that.

petenut, 2020-09-08:
Hello,
    I have been able to upload a new firmware (previously compiled .bin) and now I get the LCD12864 SILUMATOR message when I switch to Marlin, but I do not get the other options, even after moving the encoder. What should I try next?

Thank you,

J Petrie

oldman4U, 2020-09-08:
PROGRESS. Yeah👍🏻😁

So touchscreen works?

Cables?
Which kind of display is defined in Configuration.h of the mainboard?

petenut, 2020-09-08:
Yes, touchscreen works, but Marlin controls do not show up.

From SKR1.3 board firmware

Port settings:
Serial_port 1
Serial_port_2 0

Display settings:

Rep rap_discount_full_graphic_smart_controller

J Petrie

oldman4U, 2020-09-08:
Port settings are fine, because the touchscreen works.
Cables are fine, because otherwise you would not get the message.
You tried to rotate the knob, once the LCD12864 SIMULATOR shows up?

Please check in Marlin of your mainboard if the CR10 display is also activated and make sure it is not activated.

Let me know

petenut, 2020-09-08:
Hello,
   I just changed the settings to:

Serial 1 -1
Serial 2 0

After reflashing the firmware, the Greenscreen is working!

Thank you for your help.

J Petrie

oldman4U, 2020-09-08:
Greenscreen?

You make me nervous...😳

oldman4U, 2020-09-16:
Please respond or close this ticket in case the problem is solved. Thank you

petenut, 2020-09-16:
Hello Sir, yes the issue has been resolved. You may close the ticket.

Thank you,

J Petrie

oldman4U, 2020-09-16:
Petenut.

Only you can close this ticket by clicking on the Close button at the bottom of this ticket.

Please do so

Thank you

closed 2020-09-16 by petenut

oldman4U, 2020-09-16:
Thank you and happy printing

👍🏻