bigtreetech/BTT-TFT35-E3-V3.0 #32: [QUESTION]Marlin 2.0.6.1

gec100, 2020-09-01:
Marlin 2.0.6.1 introduced support for some new display types. Right now I have selected REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

is there anything new we can switch too?

thx

oldman4U, 2020-09-02:
Hi.

What kind of new features would you expect from another type of display?

gec100, 2020-09-04:
@oldman4U Right now we are using generic display type. What would I expect is better integration with Marlin. 

Don't know exactly what to expect but would like to see what's going on. They did not go through development for nothing.

oldman4U, 2020-09-04:
Not sure the type of screen would make a difference but what each screen offers today is listed on the corresponding pdf's. Now it is the "job" of the open source community to write great software for it. 

Are you a software developer or do you have other skills which could help to improve the solution?

Help is always welcome it seems.

oldman4U, 2020-09-08:
What kind of integration would you like. What are you missing in detail?

oldman4U, 2020-09-16:
Could you please respond or close this ticket. 

Thank you

oldman4U, 2020-09-16:
So lets wait for someone who knows more than me. 👍🏻

radek8, 2020-09-20:
> Promiň m8 Kdo jsi?
> 
> Položil jsem otázku, která nemá absolutně nic společného s tím, co žádáte, abych vám odpověděl.
> 
> Právě teď byl výchozí výběr REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.
> Moje otázka pro lidi, kteří ve skutečnosti vědí, o čem mluví, je, zůstaneme u toho řidiče nebo je k dispozici nová možnost?
> 
> díky

Did you not write which board you are using or what problem you have and are you waiting for an answer? You will wait a long time.

gec100, 2020-09-22:
I do not have a problem. Everything works. This is just a question. 

with the newest Marlin there is support for touch displays. Not sure what it offers. Just asking do we still use REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER or there is some other option we can use.

My board is SKR 1.4Turbo

radek8, 2020-09-22:
If we are talking about the BTT TFT35 display and similar from bigtrets, then in touch mode they have their FW which you can download here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
  In Marlin simulation mode, the BTT display supports LCD12864 display
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
or
#define CR10_STOCKDISPLAY

then LCD2004 display
the driver doesn't know because I don't use it.

So it is best for you
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

If you wanted a touch screen driver integrated into Marlin, it would mean loading a completely different FW with support for this driver into the display. But then you're on the wrong forum.
You need to visit the site that deals with the development of the driver you would like to use.

radek8, 2020-09-22:
read here

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1067

https://github.com/MarlinFirmware/Marlin/issues/18129

radek8, 2020-09-22:
If you have no other problem, close this ticket, you will make oldman4U happy :-)

gec100, 2020-09-22:
Thank you @radek8 . That was the info I needed. Nothing new but just confirming what I was thinking before. 

Closing this now 😀

closed 2020-09-22 by gec100

oldman4U, 2020-09-22:
Happy people everywhere including myself. 

Thank you radek8, well done

👍🏻

saeed3hsani, 2021-05-28:
hello to some one some where  some times . I dont have any problems but please help me to have some .. then some other felas come please and take me out of this hell pleaeaeaze

saeed3hsani, 2021-05-28:
come on I cant wait without any idea that when the problem will come please show me the way to get lost!!!

saeed3hsani, 2021-05-28:
and I warn you dont answer if you can not understand my visdommmm

saeed3hsani, 2021-05-28:
I'm sorry . it seems I've used my pills upside down again

oldman4U, 2021-05-28:
The best useless comment everrrrrrrrr. 🦸‍♂️