bigtreetech/BTT-TFT35-E3-V3.0 #34: Power draw too high, STM Chip gets hot

BinaryBandit81, 2020-09-06:
After a few hours of flashing and configuring the tft35 e3 in combination with skr pro v1.1, the power draw went way too high and the stm chip get‘s too hot to touch it.
I measured the power draw by connecting 5V to the power pins on the tft35 board itself. (There is no connection to the skr pro)
The PowerSupply shows ~6W!

What is wrong here? How can I fix this?

oldman4U, 2020-09-23:
Hi.

More than 2 weeks no answer.... so I try it.

I have exactly the same hardware, so is there a way to reproduce the issue?
What do you have connected to the TFT, besides the three cables to the SKR Pro?

Usually I have passive cooling on the mainboard and active cooling on the TFT, but I can remove this for a test.

Please let me know

Thank you

BinaryBandit81, 2020-09-24:
Hi!
Thanks for your reply. I already thought my issue was too simple that nobody bothered...

I connected the display via ext1, ext2 and/or TFT cable and also tried a direct connection to a 5V Powersupply.
Either way the power consumption was too high and chip gets hot quickly.

Besides one TFT Board blew up and does not turn on anymore...
I don’t want to repeat that on my second board.

Side note: I flashed the firmware and logos from v1engineering for the mpcnc

Is there a factory reset for the TFT Board?

Thank you!

oldman4U, 2020-09-24:
You are welcome.

I do not believe that it is simple but maybe you have a special setup which causes the issue. Not sure how to test it finally but during a print the cpu gets hot but never burned one. I know Intel CPUs can reach 105 degrees before they reduce the speed to protect themselves. 

I would cool the cpu actively with a cheap 5V fan which you can attach directly to the TFT.

oldman4U, 2020-09-26:
Another thing. What have you connecter to the TFT?

BinaryBandit81, 2020-09-29:
I didn‘t connect anything to the tft itself and I haven‘t printed/cnc‘ed anything.
As soon as I turn on the tft, the powerdraw get‘s high.
If I limit the power on the powersupply, the ftf does not turn on...

Maybe I try to actively cool the chip and try to reflash the original firmware, but I doubt that this will help.

But thank you anyways!
I will post any news if I either saved or destroyed the board... ;-)

oldman4U, 2020-09-29:
Maybe you ask your dealer for a replacement in case you believe it is broken.

oldman4U, 2020-09-30:
Please help the community and close the ticket once you do not need it anymore. Thank you

BinaryBandit81, 2020-10-02:
Seems there is no proper solution for my problem...
I stick to my old-school lcd

closed 2020-10-02 by BinaryBandit81

oldman4U, 2020-10-02:
Maybe the solution is in replacing the hardware. I can not see how software or the community can help.

Sorry