bigtreetech/BTT-TFT35-E3-V3.0 #36: Skr 1.4 turbo + tft 35 e3 can't use both exp and rs232 same time

BlackDragon2020, 2020-09-19:
If they both are plugged in, screen does not turn on on boot up. If I unplug one screen turns on and the interface for what ever cable is attached works.

Now if I let it turn on and boot up with only one cord, then plug in the 2nd with it on both interface work, untill power off then back on, then it's back screen.

Have define serial_port -1 and 0 for port 2. Also have defined the full graphic smart controller, I just added the define serial_port info under that too.

Anyone got info?

I have to use exp 3 on tft35 to make the simulated interface work.

oldman4U, 2020-09-19:
Hi. Using your mainboard, you have to connect two EXP cables which connect EXP1 and EXP2 of the TFT and the mainboard in addition to the RS232 cable.

Full Graphics Smart Controller is correct. Make sure there is no other graphics controller activated in the firmware.

Hope this helps

BlackDragon2020, 2020-09-19:
I will check tomorrow. Can you explain why I need 2 cables? also if that's the case is it still possible to run 2 screens? I was thinking of using both, just had not got the tft work yet to mess with the other.

oldman4U, 2020-09-19:
You need 3 cables.

EXP1 and 2 for Marlin emulator mode (in Marlin mainboard firmware FW Serial Port -1 and Port 2 0, as well as #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER defined as the ONLY graphics controller) and make sure you use the same baud rate for the mainboard and the TFT. Why you need 2 cables I do not know.

RS232 for touchscreen mode.

Not sure what you mean with 2 screens, but this will allow you to use Marlin emulated screen and Touchscreen.

Let me know

BlackDragon2020, 2020-09-19:
It boots up, but had an issue untill I commented out the stock ender 3 display the cr10 one. Thank you.

Do I need the #define again under the screen like this or is once okay way early in the .h file?

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define Serial_Port -1
#define Serial_Port_2 0

oldman4U, 2020-09-19:
There is no need to duplicate functions or to move them to another position. It is only important that they are defined how I wrote.

Let us know in case you need anything else. 

Please help the community and close the ticket once you do not need it anymore. 

Thank you

BlackDragon2020, 2020-09-19:
Thank you again for clearing everything up.

closed 2020-09-19 by BlackDragon2020

oldman4U, 2020-09-19:
You are very welcome 

Happy printing