bigtreetech/BTT-TFT35-E3-V3.0 #37: TFT35 E3 V3.0 not working with BTT SKR 1.4 Turbo

TheDemonPanda, 2020-09-19:
I've just installed the 1.4 Turbo board and TFT 35 V3.0 screen into my Ender 5+, but no matter how I check the cabling, the screen won't power on at all. I've connected the screen to my Ender 3 to verify that it works (it does), but nothing is working on my 5+

[image]

oldman4U, 2020-09-19:
What if you connect only RS232 or EXP1 and 2 and when you remove all the things you do not need for testing like the WIFI, SD card, external MOSFET.

TheDemonPanda, 2020-09-19:
> What if you connect only RS232 or EXP1 and 2 and when you remove all the things you do not need for testing like the WIFI, SD card, external MOSFET.

Disconnected the WIFI and SD card extender, and tried every variation of the cables. Made no difference, with and without what I assume to be the MOSFET (top of the box in the picture?) connected to the board. 
Still no signs of life from the screen

oldman4U, 2020-09-19:
So even you have only the Rs232 connected there is no sign that the display boots? No LED on the back or the display becomes illuminated?

But it works with another mainboard!?

Have you checked 5V and Ground at the TFT and mainboard are connected? Thats finally all that is needed. Test the two with a multimeter for correct voltage. 

Let me know

TheDemonPanda, 2020-09-20:
No signs of any life from the BTT screen at all with this motherboard, but worked fine with my ender 3 (said 'no printer connected', but I think that was because I only connected the EXP wire). 
I don't have a multimeter at the moment, but will pick one up tomorrow. That being said, I have no clue which bit and how to actually check for the voltage.....

oldman4U, 2020-09-20:
No Printer Connected comes from no RS232 serial connection.

You see the function of each pin on the back of the mainboard or the handbook of the same. Any sign of life of the SKR board, when you power it up? looks more like an SKR problem than a TFT problem.

radek8, 2020-09-22:
Check the 2 fuses and the correct polarity of the power supply
if it is powered, the LED next to the fuses must be lit.

oldman4U, 2020-09-23:
Any news?

Please let us know

TheDemonPanda, 2020-09-23:
Still tinkering with it. I've been given some advice on different forums that also seems promising, but it's had to delay for a short time while real life intrudes. I'll update as soon as I have anything more concrete

oldman4U, 2020-09-24:
OK.

For me this looks like an issue with the SKR board and I would focus on that first. So i would do what radek8 wrote 2 days ago.

Please let us know

TheDemonPanda, 2020-09-27:
I rechecked the wiring earlier, and though a couple of the wires did seem to be in the wrong place, nothing new has developed. I tried the screen once again on my Ender 3, and it worked perfectly. 
I've literally just picked up a multimeter, so will try to check voltages now. Otherwise, I'm thinking that this is a problem with the board, as there are no signs of life whatsoever on any screen that I attach to the SKR1.4 board

oldman4U, 2020-09-27:
There is a red LED between the current input and the fuses, which has to be lid as soon the mainboard get power. Check the polarity and the fuses, like radek8 wrote.

[image]

radek8, 2020-09-27:
Check that the fuse and the red switches are in the correct position.

[image]

TheDemonPanda, 2020-09-27:
No signs of life on the LED mentioned, oldman4u.
The fuses are all as in your image radek8

radek8, 2020-09-27:
Measure the fuse, have it not blown?

oldman4U, 2020-09-27:
You can also change the jumper from VDD to USB and try to power the board using a USB cable.

TheDemonPanda, 2020-09-27:
May be getting somewhere! The '10' fuse appears to be dead (no reaction from the multimeter). The '20' seems to work

TheDemonPanda, 2020-09-27:
Thanks for all the help so far, guys. I'm off to sort another project for the night, but I'll try and source a new fuse, then try again tomorrow

oldman4U, 2020-09-27:
👍🏻

Sounds like a possible reason for the problem.

radek8, 2020-09-27:
Don't you have a clever cousin who is an electrician and will come to lunch tomorrow? He could help you with that. :-)

oldman4U, 2020-09-28:
Now you have to tell the whole story my friend. .....🤣

TheDemonPanda, 2020-09-28:
I'm now wondering if I'm going to have a strange visit later on, because none of my cousins are sparkies, and all are useless :/

radek8, 2020-09-28:
olfman4U, unfortunately it's too long a story to tell ;-)

TheDemonPanda, If you temporarily replace the fuses, your board should be revived.
It will not heat the bed but it is enough to try.

TheDemonPanda, 2020-09-28:
I've got a new set of fuses arriving today, so hopefully it should be a more permanent fix :)

TheDemonPanda, 2020-09-28:
Though radek8, I'm also curious about the story!

radek8, 2020-09-28:
the story is here :-)
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1120

oldman4U, 2020-09-28:
Helping others has also funny sides. Radek and I had a case with a gentleman from UK. More than 100 back and forth to update the firmware and find out whats going on BEFORE we even came to the topic of the issue - no Marlin mode. Finally we both gave up and the user explained to us that he gets his cousin in the next. Day and that he has two left hands, so he has to remove the stuff before he arrives. But the user was to late and the cousin came before he could remove it - and guess what!!??

F... - he fixed the broken cable which worked ( for a still unknown reason) fine with the standard Ender3 display, but not with the new TFT.

We left the area like two wet dogs.

oldman4U, 2020-09-28:
You have been faster.

I wrote the whole story. 

🥳🥳🥳🥳🥳🥳🥳🥳🥳🥳🥳

oldman4U, 2020-09-28:
Somehow I miss Tony and the time with him😁

He was a real gentleman!!!

TheDemonPanda, 2020-09-28:
So, I ordered the wrong fuses. Wife is on route to Halfords to see if she can get the right ones 🙄

oldman4U, 2020-09-28:
By car? 😎

TheDemonPanda, 2020-09-28:
I hope so!

oldman4U, 2020-09-28:
So call her back and take the fuse from the car. I mean it is an easy
decision. Wiper or printer!?!!!🤩

TheDemonPanda, 2020-09-28:
Printer is probably worth more 😂
But if I did that, she might be inclined to try and use my car......

oldman4U, 2020-09-28:
Well. Thats right. So lets see what the outcome of this is

TheDemonPanda, 2020-09-28:
Hopefully Halfords will have the right size fuses. I think the ones I bought are the standard size for cars, those on the board are tiny

radek8, 2020-09-28:
V autě se používají obě velikosti. Alespoň v tom mém ano :-)

TheDemonPanda, 2020-09-28:
Yep. They're words

oldman4U, 2020-09-28:
> V autě se používají obě velikosti. Alespoň v tom mém ano :-)

You got the virus again my friend.

TheDemonPanda, 2020-09-28:
WE HAVE SIGNS OF LIFE!!!!

After a load of tinkering, changing the fuses, swearing, disconnecting, reconnecting, tinkering, swearing, and troubleshooting, we have a success- mostly.

I need to try and sort the firmware for the board now, which could be tricky as the SD card slot seems to be glitchy. Can I load new firmware for the board through the card slot on the screen? Or get an adaptor for the USB port and load it that way?

Unless it's a ridiculously easy fix, I wont be trying anything else tonight, I just want dinner and sleep.

radek8, 2020-09-28:
The SD card slot may not be defective.
if you have a choice in Marli
#define SDCARD_CONNECTION LCD
The card will not be available for printing, but you can run flash FW from it

oldman4U, 2020-09-28:
First Tony's cousin and now the Pandas wife....

Seems we are not needed anymore😬

TheDemonPanda, 2020-09-29:
Tried updating the firmware this morning, and the SD card slot on the board is non responsive. The LED for it lights up to show a connection, but there doesn't appear to be any communication between it and the screen just says 'Read Onboard SD card error' whenever I select that option. I've tried a few different cards, all with the same result.

Also, the screen keeps dimming to the point it's unreadable. Any clues?

radek8, 2020-09-29:
You set the dimming of the display in the Config.ini file
Save it to the SD card and restart the display.

The SD card on the Marlin board will be available if you have the value enabled in Marlin
#define SDCARD_CONNECTION ONBOARD
SD card should not be larger than 32GB (preferably less)
and should be in FAT32 format

oldman4U, 2020-09-29:
The ability to use the SD card for firmware update is always available and completely independent of any settings in Marlin or TFT. I would recommend to use the SD card which came with the device to update the firmware. The 128MB are more than sufficient.

radek8, 2020-09-29:
TheDemonPanda  talks about the SD card on the board, and it's available depending on Marlin's settings.

oldman4U, 2020-09-29:
Marlin and TFT settings if you want to print. I only talked about firmware update and the King of Pop.

oldman4U, 2020-10-01:
Hello Panda. Any news?

Can the sd card be used to upload new firmware?

TheDemonPanda, 2020-10-01:
Doesn't appear to be working. The LED on the board lights up when I put the card in, but nothing happens. I've tried a couple of cards, and nothing is responding. Can I use the USB to update the firmware?

oldman4U, 2020-10-01:
You have to power up the mainboard with the <8GB SD card (which has the firmware.bin file on it) already inserted. 

My SKR 1.4 shows no activity while updating but after 30 sec you can remove the card to see if the file extension changed to CUR.

TheDemonPanda, 2020-10-01:
Just tried it again, left it for a few minutes, and checked the card. Still showing as .BIN
May download the arduino software and see if I can do it straight from the PC

oldman4U, 2020-10-01:
It will only update from SD Card.

Do you know the reason for the blown fuse?

TheDemonPanda, 2020-10-01:
Not a clue why the fuse originally blew. 
Have I got any other option for doing the firmware if the SD reader isn't working? At this point, I'm thinking of just changing to another board entirely (I've got a couple of spare SKR mini E3 1.2's and some splitters)

oldman4U, 2020-10-01:
Not that I know. Card is the only way as far as i know.

At least you could test the display with the spare board.

TheDemonPanda, 2020-10-01:
The display now seems to be working properly, though won't show anything in the Marlin mode, I assume because I can't flash the firmware.

oldman4U, 2020-10-01:
For Marlin you have to connect the two 10pin ribbon cables and you have to define the right reprap display in Marlin.

TheDemonPanda, 2020-10-01:
I've got the two ribbon cables and the RS232 connected to the screen

oldman4U, 2020-10-01:
Correct EXP 1 and 2

What do you seewhen you switch to Marlin Mode?

TheDemonPanda, 2020-10-01:
Absolutely nothing, blank screen. Touchscreen mode works well enough, except it's irrelevant as the firmware is wrong 😂

radek8, 2020-10-01:
An SD card is not the only requirement for a flash drive

https://www.youtube.com/watch?v=7O_ZOdnM0SU

radek8, 2020-10-01:
WF upload via also USB does not work?

[15 more replies]

closed 2020-10-12 by TheDemonPanda