bigtreetech/BTT-TFT35-E3-V3.0 #38: Tft35 e3 v3 not turning on

kabone, 2020-09-19:
I have a skr mini e3 2.0 and tft35 e3 v3. It was working good. After a reboot of my printer the screen is not turning on. I ordered a replacement that one also is not turning on. If i flip the tft pins it turns on with a white line across. If i remove the tft pins and llugin to ext1 it powers on and looks correctly. Anything i can do to overcome this?

oldman4U, 2020-09-20:
What do you mean with remove the tft pins?

You have to connect the RS232 cable on the TFT and the mainboard and EXP on mainboard to EXP3 of TFT. But you wrote this has been working before so you know this....

kabone, 2020-09-21:
> What do you mean with remove the tft pins?
> 
> You have to connect the RS232 cable on the TFT and the mainboard and EXP on mainboard to EXP3 of TFT. But you wrote this has been working before so you know this....
I mean remove the tft wire. It was working, but ive since ordered a replacement to see if it was in fact the tft35 it turned out its an issue with the skr mini e3 v2.0. Both lcds worked when i hooked up the new board. The new lcd and old lcd did not work with the old skr mini. Ill go ahead and marked this closed.

oldman4U, 2020-09-21:
Problem is solved!!??

This would be great 

Thank you for closing the ticket 

Happy printing

kabone, 2021-03-15:
Well I had ordered another one after giving up with it. The new one just now did the same... Same exact situation Has been working for a year, turned it off after a print to turn it back on to a blank screen..

radek8, 2021-04-25:
If you still have a problem, send Marlin configuration files and connector wiring photos.
I'll see if anything comes to mind

kabone, 2021-04-25:
> If you still have a problem, send Marlin configuration files and connector wiring photos.
> I'll see if anything comes to mind

I reached out to the facebook support and sent videos of all this. They issued me 2 new boards. Everything worked right away with the new board and tft35 e3 v3

closed 2021-04-25 by kabone