bigtreetech/BTT-TFT35-E3-V3.0 #39: No Sounds coming from Speaker on Display

amodernstitch, 2020-09-20:
Any tricks for being able to turn sound on? I have no sounds coming from touch display. 
I have a BTT SKR Mini E3 V2.0 and the BTT-TFT35-E3-V3.0 installed on my Ender 3 Pro

oldman4U, 2020-09-20:
Hi

Have you checked if Sound is on under the screen settings of the TFT?

amodernstitch, 2020-09-20:
yes, I have toggled sound on and off on the touch screen. 
I believe I have the latest firmware for the TFT board
What needs to be uncommented in Marlin 2.0? 
What is another way to test the buzzer/speaker?

oldman4U, 2020-09-20:
With sound on you should hear a click every time you press a button. If this is not the case the hardware is broken. I replaced one already with a spare buzzer I had hanging around.

amodernstitch, 2020-09-20:
I don't hear anything. I have the original screen buzzer from the Creality Ender 3 Pro. Is it as simple as using a soldering iron to get it off the old board and replace the ?defective? one?

oldman4U, 2020-09-20:
Broken. 

Yes it is. I added a cable so I could move it a bit away from the front.

[image]

amodernstitch, 2020-09-20:
Wondering if I should get a new display from BTT since the board came this way. I don't think I should have to break out the soldering iron to fix something that should have arrived in working order.

oldman4U, 2020-09-20:
I agree. Contact the dealer you bought it from.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-09-20 by amodernstitch

oldman4U, 2020-09-20:
Thank you and hope you find a good solution for the buzzer.