bigtreetech/BTT-TFT35-E3-V3.0 #40: there is no communication with the printer and bltouch or abl leveling

valerox84, 2020-09-25:
I have checked the serials and even tried others and the result is still the same, the baudrate is the same in tft and marlin, the printer responds but the screen says that there is no communication with the printer.
Also in the leveling section, before abl or bltouch came out to level in the last firm of the tft with marlin 2.0.6.1 has disappeared

marlin 2.0.6.1
tft e3 v3. 0.26.x
skr e3 dip v1. 0

valerox84, 2020-09-25:
I just verified that it is not a failure by marlin, it is a failure of the tft firm, even when you want to update the firm and others not, tried with 5 different sd, but the printer is not connected and the leveling icons of bltouch

oldman4U, 2020-09-25:
Hi.

I have the same mainboard and TFT working fine since almost a year.

Please attach pictures which show the cabling and the Info screen on the TFT.

Thank you

valerox84, 2020-09-25:
Hello, first of all thank you very much for communicating with me, I have managed to correct the problem, the reset cable was disconnected and it was the one that was giving me the faults, now it works perfectly, thank you very much for all the work you do.
a greeting

Obtener Outlook para Android<https://aka.ms/ghei36>

oldman4U, 2020-09-25:
Hello valerox84.

It is a welcome variety to have such a kind person asking questions;-)

Please help us to keep this system usable and close this ticket by clicking on Close at the bottom of this page.

Thank you and happy printing

closed 2020-09-27 by valerox84