bigtreetech/BTT-TFT35-E3-V3.0 #41: TFT35 won't turn on with SKR 1.4 (need to unplug and re-plug EXP3 to work)

skippy3111, 2020-10-04:
My TFT35 won't turn on with my printer. I need to unplug the EXP ribbon cable and plug it back in, then it works fine (in either LCD or TFT mode) till I power off my printer again. 

Some notes: 
 - Printer is an Ender 3 pro with an SKR 1.4 Turbo. Running Marlin 2.0.x. 
 - I have the EXP1 port on the mainboard connected to EXP3 on the TFT35. 

Any suggestions on how I can get it to power on with the printer without having to unplug and plug it back in?

Thanks!

oldman4U, 2020-10-04:
Where did you saw that EXP cabeling?

Roughd1433, 2020-10-04:
It should both exp cables to the same ports in the board and tft. Then the tft wires from the screen to the board. You need to make sure you match the 5v pins on the cable to the correct one on each end as some cables have the wires swapped when they were made.

oldman4U, 2020-10-04:
It is important to know how a user can connect the cables so wrong. In this case the number of cables and the connector is wrong and somewhere the user saw this as the right solution.

oldman4U, 2020-10-05:
But maybe this is one of the typical "first issue ever", where the user was only able to open the ticket, but not to close it...

@skippy3111 What is the status of the reported issue?

skippy3111, 2020-10-05:
Hi guys, sorry got busy yesterday and haven't had a chance to try the 2nd
cable as suggested, as well as the other suggestions.

FYI I power on with only the TFT cable plugged in, it works in TFT mode
without any issues. So I don't think it's the TFT wire issues mentioned?

Let me refine the the problem description... if I want to use it in LCD
mode I need to plug in the LCD (EXP) cable after boot. If it's plugged in
at boot, even TFT mode is a blank screen until the cable is unplugged.

I'll take some pictures of the my wiring later today

Thanks!

oldman4U, 2020-10-05:
Again. Where do you get the cable information from?

oldman4U, 2020-10-05:
Connecting the cables wrong can also destroy your hardware, mainboard and/or TFT.

skippy3111, 2020-10-05:
Hi guys, was busy with work all day. Just got a chance to have a look now. 

> Where did you saw that EXP cabeling?

@oldman4U I was following [Teaching Tech's video on SKR 1.4 installation](https://youtu.be/-Gdk0wHg51w?t=975) and [his video on the TFT35](https://www.youtube.com/watch?v=l1Putn10_Ek&t=270s). 

> It should both exp cables to the same ports in the board and tft. Then the tft wires from the screen to the board. You need to make sure you match the 5v pins on the cable to the correct one on each end as some cables have the wires swapped when they were made.

@Roughd1433 Tried this setup (see below). I also traced the 5v line (white/black on the rainbow ribbon), the red line on the grey ribbon, and the wires on RS232 cable and compared it to the pin-outs for the SKR1.4 and the TFT35. Seems to be all okay. 
[image]

With this setup the screen would power up with the printer and work in TFT mode, but the LCD mode would not work. To make this cabling work in LCD mode, should I change anything in the firmware?

Here's some other combinations I tried: 
**Case 1** (SKR EXP1 to TFT EXP3) - the screen powers on with printer and works in LCD mode. 
[image]

**Case 2** (SKR TFT to TFT RS232) - the screen powers on with printer and works in TFT mode. 
[image]

**Case 3** (SKR EXP1 to TFT EXP3 & SKR TFT to TFT RS232) - screen does not power on with printer. If I unplug and re-plug either cable it will continue to work in either mode as desired (until I power off the printer). 
[image]

> Connecting the cables wrong can also destroy your hardware, mainboard and/or TFT.

so far no issues 😅🤞

Thanks for the help guys!

oldman4U, 2020-10-06:
Hi. 

Checked the movies and our friend from down under explains it right. 

[image]

and
[image]

So connecting RS232 for touch screen support AND EXP1 and EXP2 of the mainboard and the TFT for Marlin simulation. 

In Matlin ONLY REPRAP DISCOUNT FULL GRAPHICS SMART CONTROLLER must be activated. CR10 has to be deactivated by adding // in front. 

Let us know

skippy3111, 2020-10-07:
> So connecting RS232 for touch screen support AND EXP1 and EXP2 of the mainboard and the TFT for Marlin simulation.
> 
> In Matlin ONLY REPRAP DISCOUNT FULL GRAPHICS SMART CONTROLLER must be activated. CR10 has to be deactivated by adding // in front.

@oldman4U ok so the screen powers on with the printer now. Works fine in TFT mode. In LCD mode, the knob's push button registers, however turning it doesn't do anything. 

FYI Here's the configuration files that I compiled with [Marlin-bugfix-2.0.x.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/5338146/Marlin-bugfix-2.0.x.zip)

I didn't have a lot of time to dubug a lot today though. I'll give it another shot tomorrow evening. 

I'll start with compiling a new firmware file with a fresh set of Marlin configuration files with only the instructions you noted above, and will go from there. 

Will keep you posted. Thanks!

oldman4U, 2020-10-07:
What do you mean with turning doesn’t do anything?

skippy3111, 2020-10-07:
I can't scroll through the menus with the wheel. I can click once to enter
the menu screen, and click again to go back to the info screen, but can't
switch to any other menu option.

skippy3111, 2020-10-09:
(been swamped with work, will try to make some time tomorrow)

oldman4U, 2020-10-09:
When you are in the mode selection area, can you switch between the two modes using the knob?

skippy3111, 2020-10-10:
> When you are in the mode selection area, can you switch between the two modes using the knob?

nope. no response from turning the knob.

oldman4U, 2020-10-10:
Looks like a hardware issue to me. Had a broken encoder too. Live without Marlin or replace it.

skippy3111, 2020-10-10:
Yeah I was afraid of that. I wonder if I damaged something when unplugging
and replugging the cables.

Will try to revert back to the previous firmware with the cr10 option
enabled.

Too darn busy with Thanksgiving though.

skippy3111, 2020-10-11:
Success! Had a bad cable and it worked when I switched it out. 

So to recap, solution is to enable:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

then disable:
//#define CR10_STOCKDISPLAY

and connect EXP1 to EXP1, EXP2 to EXP2. 

Thanks for the support @oldman4U and @Roughd1433!

closed 2020-10-11 by skippy3111

oldman4U, 2020-10-11:
And finally you found the reason for the problem!

👍🏻

Happy printing