bigtreetech/BTT-TFT35-E3-V3.0 #42: Stuck on Touch Screen Calibration (please click on the red dot) after flashing firmware

pdsavas, 2020-10-08:
I have been trying to get a brand new screen working. I flashed the firmware, but when I get to the Touch Screen Calibration, the TFT is not responding to touch, so it sits at waiting for the red dot in the upper left corner of the screen to be pressed.

I'm fairly new to this so any help would be appreciated.

Thanks,
pete

oldman4U, 2020-10-09:
Hi. Are you using pre or self compiled firmware. SD card size and format. Which files are on the card. Whats the name of the files before and after update?

pdsavas, 2020-10-09:
I've tried using pre-compiled versions of v25 and v26. The SD card is a 1GB and formatted in FAT32.

The files on the card for v25 are: BIGTREE_TFT35_V3.0.25.bin (109,312 bytes both before and after update) before update and BIGTREE_TFT35>V3.0.25.CUR after update. Also, the entire TFT35 directory. The TFT35 directory gets renamed TFT35.CUR after the update.

The files for v26 are: BIGTREE_TFT35_V3.0.26.1.bin (162,584 bytes both before and after update) before update and BIGTREE_TFT35_V3.0.26.1.CUR after update. Also, the entire TFT35 directory. The TFT35 directory gets renamed TFT35.CUR after the update.

I even tried using the files from your fork (BIGTREETECH-TouchScreenFirmware-Oldman4U) with no success.

With all three, I used the files from the Unified Menu Material theme.

Please let me know if you need more information.

oldman4U, 2020-10-09:
You have to use E3 based firmware version and update TFT35 folder as well
as config.ini like described on main page.

pdsavas, 2020-10-09:
When I tried using the E3 based firmare, I got a screen, after the update, that said:

Found outdated data:

Config: Update required

Insert the SD card with the required
files and press the reset button
to update.

oldman4U, 2020-10-09:
Thats right. And the error is because you are not reading.

pdsavas, 2020-10-09:
Please then, instead of being condescending, tell me what I am missing. As I said, I'm VERY new to all of this and trying hard to figure it out. I've been reading and watching videos for weeks now, and yet I'm still missing something basic, apparently.

oldman4U, 2020-10-09:
Your card is fine, using E3 firmware is correct and when you read the ReadMe and the error you have seen on screen you will also know what to do. 

Config: Update required

Insert the SD card with the required files and press the reset button to update.

pdsavas, 2020-10-09:
Okay. I've gone back and re-read the pages (both the main and the config instructions). I THINK I have it figured out and am trying to flash the display once more (this time with a config.ini file on the card). I'm using the BIGTREE_TFT35>V3.0_E3.26.bin (224,960 bytes) from the Unified Menu Material folder of the Master build.

And...success with the TFT!

I apologize for being frustrated. I've actually been building this printer from scratch (after having an Ender 3 for a while) using an Ender Extender XXL kit as the base frame. I've had to buy all the parts, and make cables needed that didn't come with the kit, so it's been a two-month-long learning process for me; I haven't even been able to test the hardware to see if it's going to work properly before trying the hot end and finally printing. Getting the screen working was the last thing I needed to do before actually checking if the motors will work, if the BLTouch will work, etc. It's been nerve-wracking at best; frustrating at worst.

Thank you again.
pete

oldman4U, 2020-10-09:
Hi Pete.

I completely understand you and it is great you could get the TFT working.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-10-09 by pdsavas