bigtreetech/BTT-TFT35-E3-V3.0 #43: Touchscreen doesn’t respond and screen is NOT touching the frame where it’s mounted in.

jlogin2002, 2020-10-11:
Touchscreen doesn’t respond and screen is not touching the frame where it’s mounted in.

First time I swich on, system shows me a calibration screen but I could not do nothing at all. I powered off, powered on and touchscreen modo dos not work but emulation .

All ribon cables are ok, marlin configuration is ok I can operate my printer in emulation mode.

I look forward to hearing from  your news soon

Thank you

oldman4U, 2020-10-11:
Hi.

This is a brand new TFT or has it been working before?
Which mainboard?
What exactly is connected on which connector of the TFT?
Which firmware and have you updated the firmware already?
Does it work with only the RS232 cable connected?

jlogin2002, 2020-10-11:
I close the issue into this branch because I have opened in the right place. I respond into the proper thread

closed 2020-10-11 by jlogin2002