bigtreetech/BTT-TFT35-E3-V3.0 #44: No display while connected to either port

dustybandago, 2020-10-22:
I went to turn on my printer and everything turns on except for the display. when I connect the ribbon cable to the display I can get a beep when I press down on the button but that it. nothing happens with the serial port at all. I also tried the reset button with no results.

oldman4U, 2020-10-23:
Hi. 

Which mainboard are you using?

dustybandago, 2020-10-23:
skr mini e3 v2.0

oldman4U, 2020-10-23:
So nothing even you have only serial cable connected?

Can you send pictures of front and back of the screen while connected and turned on.

dustybandago, 2020-10-23:
I connected the OEM display from my ender 3 and it works so I know its not the main board

dustybandago, 2020-10-23:
[image]
[image]

oldman4U, 2020-10-23:
The display gets power....

Can you hear a sound when you touch the screen while it is turned on.

Looks like a broken backlight to me.

This is how the bottom of a working screen looks like.  

[image]

dustybandago, 2020-10-23:
there is no sound when I touch the screen. I don't even get the beeps to swap over to marlin mode. the backlight isn't coming on either. 
I think I'm just going to toss it up as a loss and take it as an opportunity to get the TFT 70.

oldman4U, 2020-10-23:
You should contact your dealer if it is new. But this unit seems to be broken 😔. Sorry

dustybandago, 2020-10-23:
it ok, thank you for your help

closed 2020-10-23 by dustybandago

oldman4U, 2020-10-24:
You could do the ultimate test by cutting a USB cable and connecting the red and black strip to 5V and GND to the serial port of the TFT. It has to start normally.

dustybandago, 2020-10-24:
I was able to get a replacement but I might try that too.