bigtreetech/BTT-TFT35-E3-V3.0 #45: Screen tells me no printer attached with SKR Mini E3 V2.0

rasimif, 2020-11-04:
Screen is connected to said board at 115200 baudrate, there's nothing else connected to the board except the screen and power supply. Any ideas what I'm doing wrong?

oldman4U, 2020-11-04:
Which baud rate do you have defined in Marlin of the SKR?

Attach pictures showing the connection from SKR to TFT.

Is this a new setup or has this been working before?

Any spare parts you could use to try different setup?

The screen shows the touchscreen UI as soon as it gets power.

rasimif, 2020-11-04:
Thanks for your swift reply. 
The baudrate is defined as 115200, this is meant to be an upgrade to an old printer, changing board + screen. 
Old board and screen are from an Anet A6 running Marlin. The reset button on the screen works and resets screen + board.
Simplify 3D gets a connection to the board over USB.
[image]
[image]

closed 2020-11-04 by rasimif

oldman4U, 2020-11-04:
Cable wrong....

Would be nice to let me know.