bigtreetech/BTT-TFT35-E3-V3.0 #48: TFT35 E3 V3 on SKR Mini E3 V2 Touch panel not working

nor1bie, 2020-11-14:
It was working until I got the latest and greatest. I been compiling board firmware and flashing TFT firmware like there is no tomorrow, to no avail. The hardware did work prior to the described above. The LCD simulator does work, but I find it cumbersome to quickly move position(s). 
What could have happened or got stuck on the road to firmware upgrade?
I can't think of anything else that I could do.
Unfortunately, I can't provide much more technical background as there is not much to tell. Working one day, upgrade, no more working touch panel!
Any suggestions?

Cheers

oldman4U, 2020-11-14:
Explain not working.

oldman4U, 2020-11-16:
It would be great if you could answer or close the ticket in case you found the reason for the problem.

Thank you

nor1bie, 2020-11-16:
After several hours of troubleshooting and re-flashes, it seems the kit I received is faulty, so I will return it for replacement.
For whatever reason, **I assume**, the RS232/TFT connection went out of business. This connection is needed to communicate the touch panel. From what I can see, there are malformed serial packets between the controller and display. Didn't investigate which of the two is responsible for messed up packets.

closed 2020-11-16 by nor1bie

oldman4U, 2020-11-16:
Ok.

Thank you