bigtreetech/BTT-TFT35-E3-V3.0 #49: Where is the firmware?

artiface, 2020-11-15:
I do not see any firmware files for this device in this github?

oldman4U, 2020-11-15:
There is a firmware repository.

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2020-11-16:
Have you found the firmware?

Please let me know

oldman4U, 2020-11-24:
Please help the community and close the ticket once you do not need it anymore. Thank you

doug-scott, 2020-12-24:
Where is this firmware located?  Simply saying "There is a firmware repository" is only stating the obvious.  I know there must be a firmware repository, but, I do not know where.

radek8, 2020-12-25:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

oldman4U, 2020-12-26:
artifice Have you found the firmware?

oldman4U, 2021-01-09:
@doug-scott This is the reason why it is so hard to help. There are many "I open a ticket for the first time" users, which are not able or willing to make two clicks more, but it is easier for them to open the ticket and never close it after that. The alternative would be to give no answer at all, which would end up like this: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues

@artiface Respond or close the ticket

artiface, 2021-04-09:
thanks to radek8 for actually posting the link which was helpful, unlike the other commenter just telling me to close the ticket over and over.

closed 2021-04-09 by artiface