bigtreetech/BTT-TFT35-E3-V3.0 #50: Hi. Which sd card and which firmware. Has the screen working before or is it a new installation?

chadumisprime, 2020-11-22:
Hi. Which sd card and which firmware. Has the screen working before or is it a new installation?

_Originally posted by @oldman4U in https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/33#issuecomment-687174990_

I’m having this same issue

oldman4U, 2020-11-22:
Which issue?

More information could help. Like firmware version, pictures of the cabling, problem description...

chadumisprime, 2020-11-22:
I have attached a few photos and a video of the issue. I’ve updated to marlin 2.0.7  I also have a Bltouch connected. 
When I turn the ender 3 pro on, the screen comes on the shuts off within 10 seconds. This also happens when I plug the original printer screen in. 
[image]
[image]

oldman4U, 2020-11-22:
Sounds like the tFT does not get enough power.

You can verify if the TFT works by cutting a USB cable and connect the red wire to 5V and the black wire to GND of the Power connector. Doing so the TFT should power up and stay like this. Disconnect all other cables from the TFT for this.

oldman4U, 2020-11-27:
Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-11-30 by chadumisprime

radek8, 2021-04-25:
According to the photo, I assume that the problem was in the incorrectly connected connector from BL-Touch?
It is so?