bigtreetech/BTT-TFT35-E3-V3.0 #51: SKR E2 V2 and bigtree tech e3 v3 tft screen wifi not working. esp 8266

Talha909, 2020-12-01:
I had SKR E2 V2 and bigtree tech e3 v3 tft screen. I also have ESP8266. I configure the ESP8266 and installed the esp into my tft screen. I connect the tft with the wifi. But i cant able to move my printer. Do i have to make any change in marlin.

oldman4U, 2020-12-01:
Hi

Please send a picture which shows the module and the TFT.

Talha909, 2020-12-02:
[image]
[image]
 I did not change the any thing in Marlin. I put the esp 8266 on the back of tft. I got the display of esp8266 on my screen. And lot errors comes on the lcd screen.  But i cant able to control the printer from the lcd

oldman4U, 2020-12-02:
Please send a picture of your tft and esp device. I did not change anything in Marlin.

closed 2020-12-12 by Talha909

oldman4U, 2020-12-12:
Does it work?

Talha909, 2020-12-12:
I simply purchased the new esp01s and its works good. I did not whats the problem with the old one. bout are same.

oldman4U, 2020-12-12:
👍🏻