bigtreetech/BTT-TFT35-E3-V3.0 #55: Blue Screen

pinzilla, 2021-01-03:
I was going from the touch screen to the dial screen and the screen went solid blue, I tried to reset it but that didn't work so I tried to unplug cables that didn't work either. I put old screen back on and it works. Any ideas?

oldman4U, 2021-01-05:
Dial screen you mean Marlin emulation screen?

pinzilla, 2021-01-05:
Yes sorry the marlin screen (I'm a newbie to this stuff) I did get the touch screen to come on but says cant connect to printer and the marlin screen is just A blank blue screen.

oldman4U, 2021-01-05:
Please go to the firmware repository and read the readme on the fron page.

radek8, 2021-01-10:
You have to write which Marlin board you use, what cables you connect (preferably a photo).
Attach the Marlin configuration.
You must set the same communication speed.

pinzilla, 2021-01-10:
Thanks for all your help, I have read everything and tried everything people have suggested but with no luck, The only thing I have learned was that a whole lot of people have issues with this. I got two small prints with this and then everything went haywire and now it is not worth the time and aggravation to spend on this. I have better things to do like go and clean up all the dog poop from the yard and clean all the gutters out etc. etc. I will just set it out at 50 yards and dial in my scope.

oldman4U, 2021-01-10:
Ok. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-01-12 by pinzilla