bigtreetech/BTT-TFT35-E3-V3.0 #56: "no printer attached" error on skr e3 turbo

DannyGi82, 2021-01-04:
I've installed todat the skr e3 turbo on my ender 3 v2. The pack from amazon is comprensive with tft35-E3 v3.
I've follow the pin diagram and connect the tft cable to the board, update the firmware with last pre compiled, set the baudrate for both to 115200, but the error is still in place.
The port configuration is:
`#define SERIAL_PORT -1
#define SERIAL_PORT_2 0`

Uncommented the `#define CR10_STOCKDISPLAY` and `#define REPRAP_DISCOUNT_SMART_CONTROLLER` and `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

Marlin emulation work perfectly.

How i can fix this? Thank you

oldman4U, 2021-01-05:
Only CR-10 Stockdisplay and serial port 2 and -1.

DannyGi82, 2021-01-05:
> Only CR-10 Stockdisplay and serial port 2 and -1.

Done. Same error message

oldman4U, 2021-01-09:
Broken hardware? Most likely the cable.

radek8, 2021-01-10:
Olodman4U
-1 is a USB port
set
 #define SERIAL_PORT 2
 #define SERIAL_PORT_2 - 1

DannyGi82, 2021-01-10:
> Broken hardware? Most likely the cable.

No, the tft cable have rx and tx pin inverted.
Push out this pins from block and put in in right position and voilà the tft35 magically work!

closed 2021-01-10 by DannyGi82

oldman4U, 2021-01-10:
So you mean that two of the five strings of the RS232 cable have been twisted and this cable came with the TFT?

DannyGi82, 2021-01-10:
> So you mean that two of the five strings of the RS232 cable have been twisted and this cable came with the TFT?

Yes, exactly