bigtreetech/BTT-TFT35-E3-V3.0 #57: Speed percent changes by itself.

polacoloco, 2021-01-05:
Often I have to select the speed percent and change it back to normal (100%).  Something is randomly putting in other percentages like 104%, 242%, 123% and so on. I have to be really careful to check this and not starting a print with the speed set to high. Really annoying!

oldman4U, 2021-01-07:
Do you see two values when you change a setting like speed, flow or fan?

DerInternets, 2021-01-07:
Same issue here - menu shows xxx/xxx - so current / expected value. I also get completely random speed settings from time to time. Last value was nearly 400% - the noises my double z-axis motors made when trying to move 50mm upwards were horrible.

oldman4U, 2021-01-08:
Maybe a video showing the issue would help.

sarvenn, 2021-01-09:
I had the same issue whenever I change the speed manually. Even while changing the percentage value to %100 manually it jump again to a different value.

oldman4U, 2021-01-09:
Load the master firmware and let us know.

polacoloco, 2021-01-10:
Im no expert on this stuff. Where can i find the master firmware? Ive just downloaded the files from this page.

radek8, 2021-01-10:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

polacoloco, 2021-01-10:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

This is the files that are currently in my tft. I had the same issue with the stock firmware on the tft.

oldman4U, 2021-01-10:
Please make a movie which shows the issue and attach it here.

radek8, 2021-01-10:
The speed changes when you turn the encoder wheel ....
Am I right? :-)

radek8, 2021-01-10:
This is a well-known problem.
If you only use EXP3 to connect Marli and TFT, you have this problem.
This could also be addressed, but Marlin will need to be modified and another wire connected. I haven't tried it, but it should work.

oldman4U, 2021-01-11:
By itself is something different for me....😁

oldman4U, 2021-01-12:
Any news?

Please let us know or close the ticket.

Thank you

polacoloco, 2021-01-13:
> This is a well-known problem.
> If you only use EXP3 to connect Marlin and TFT, you have this problem.
> This could also be addressed, but Marlin will need to be modified and another wire connected. I haven't tried it, but it should work.

I use two cables to connect the tft to my skr mini e3 v2.0. The ones that came with the tft. How should I modify marlin?

oldman4U, 2021-01-13:
Please upload a movie showing the issue.

radek8, 2021-01-13:
Yes, you are using 2 cables.
1pc - 10pin for Marlin mode
1pc - 5pin for touch mode.
you have a SKR mini V2.0 board - important information!
Can you confirm that the problem is with changing the print speed if you turn the encoder knob?

polacoloco, 2021-01-13:
Yes

polacoloco, 2021-01-13:
Yes, now I noticed. F.ex when I increase the nozzle temperature with the knob, when I go back to the main screen the print speed has changed.

radek8, 2021-01-13:
No video required. Oldman4U finds some other erotic video on Youtube :-)

radek8, 2021-01-13:
Now write me what type of display you have.
The problem described is a HW problem in the encoder connection.
This problem can be solved by SW repair.
Write me what version of Marlin you are using (must be newer than 25.10.2020)
I also need to know which pin you don't use on the marlin board and maybe use it.

polacoloco, 2021-01-13:
#
[image]

pins not being used I don’t know. Got no extra fans or stuff like that connected. Only the bltouch.

radek8, 2021-01-13:
Try this adjustment:
1. connect pin PC13 on SKR to pin PC6 on TFT.

[image]

2. Edit the file "pins_BTT_SKR_MINI_E3_V2_0.h" according to the picture

[image]

#if ENABLED(CR10_STOCKDISPLAY)
   #define BTN_ENC_EN        PC13  // Detect the presence of the encoder
#endif

3. Compile and upload the new Marlin.

polacoloco, 2021-01-14:
Thanks! I will try this as soon as possible and get back to you with the results!

radek8, 2021-01-14:
@polacoloco, beware your version of Marlin does not have a patch implemented. 
Use Marlin "Nightly" Builds = bugfix-2.0.x

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

demandzm, 2021-01-15:
I have the same issue with a tft35 v3 connected to a skr1.4 turbo with ext1 ext2 and serial. Does this fix still apply with 3 cables?

radek8, 2021-01-15:
The correction was implemented in the 10th month of 2020.
It uses a free pin on the EXP1 connector to detect the mode
Therefore, if you have the current FW on the TFT and the EXP1 + EXP2 connector is used for the connection, you should not have this problem.
If you have this problem, it means that you do not have an up-to-date Marlin.
Use Marlin "Nightly" Builds = bugfix-2.0.x
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

oldman4U, 2021-01-20:
Please help the community and close the ticket once you do not need it anymore. Thank you

polacoloco, 2021-01-20:
> Please help the community and close the ticket once you do not need it anymore. Thank you

I will. I promise. Don’t you worry bout that. My printer is currently being rebuilt so it might take a couple of days. 
Thank you

oldman4U, 2021-01-20:
Perfect

Happy printing

craftyham, 2021-02-06:
> 
> 
> Often I have to select the speed percent and change it back to normal (100%). Something is randomly putting in other percentages like 104%, 242%, 123% and so on. I have to be really careful to check this and not starting a print with the speed set to high. Really annoying!

I noticed this as well. I have yet to pin point when it happens; & what was done last.

oldman4U, 2021-02-06:
In case you see only one value and not two (current/defined) you have an
old firmware of the TFT.

Update and you will see two values and the problem will be gone.

polacoloco, 2021-02-06:
> I noticed this as well. I have yet to pin point when it happens; & what was done last.

It’s when you use the encoder knob to change other values such as nozzle temp, it actually change the speed percent in the background. Stop using the knob and the speed will stop changing. Try it

oldman4U, 2021-02-06:
Correct. Using latest version this has been fixed for mainboards connected
via EXP 1and 2 but can not be fixed in case of EXP 3 connected mainboards.

radek8, 2021-02-07:
@polacoloco have you tried my proposed edit?

closed 2021-02-15 by polacoloco

phongshader, 2021-09-17:
I realize this has been closed for a while but I having this exact issue. I'm running Marlin bugfix-2.0x 09/11/21 and BIGTREE_TFT35_V3.0_E3.27.x.bin from the same day.
when I start the printer the speed is 100% when I load gcode the speed multiplier jumps to anything from 150% to 350%. I can reset the multiplier to 100% but if I load the same gcode the multiplier will change to some other greater speed multiplier. The multiplier is NOT in the gcode.
Is there a fix for this, other than not using the knob to adjust settings?

oldman4U, 2021-09-18:
You can fix it by buying a mainboard which connects via EXP1 and 2 or disable Marlin on yours.

radek8, 2021-09-18:
@phongshader .
You didn't specify what Marlin board you were using.
This is quite important information.

It is a well-known problem that has several solutions.
1. You stop using the encoder in touch mode
2. disconnect the EXP3 cable and you will not be using Marlin mode
3. you buy a board that has ECP1 and EXP2 connectors
4. In Marlin, disable the return to the status screen after the set time has elapsed. and after turning on the printer, just one click on the encoder and then you can rotate the encoder and the speed will not change. Be careful not to click in touch mode anymore.
5. Use the hack above.

phongshader, 2021-09-19:
Sorry, the board is a skr mini e3 running Marlin bugfix-2.0x 09/11/21. So there is no solution that keeps the full functionality of the touchscreen/marlin mode? When you say the hack above are you referring to https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/57#issuecomment-759822436

radek8, 2021-09-19:
Yes. 
There are five solutions. I've described them all.

phongshader, 2021-09-19:
Thank you...solutions...sort of, 3 is not really a solution, change the horse to fit the shoe?, 4 "Be careful not to click in touch mode anymore." 5 is really the only solution, unless I'm misunderstanding how 4 works. Nevertheless I can work around this with a startup script in the slicer.

radek8, 2021-09-19:
Never click the encoder in touch mode. 
You can only spin the encoder if you can get the pulses of the coder to keep the Marlins from.

radek8, 2021-09-19:
How would you like to handle this in the startup script?

phongshader, 2021-09-19:
simple,  m220 s100 so it gets reset to 100 regardless of what the encoder says...right?

radek8, 2021-09-19:
Yes, when you start printing you will have a speed of 100%, but you will find that you need to tune the child step or temperature. As soon as you touch the encoder, you also change the print speed.
This is not a good solution.

radek8, 2021-09-19:
The M220 S100 command only sets the value.
It does not block input from the encoder

oldman4U, 2021-09-19:
If this would work it would only make sense if you would be able to change the print speed using the encoder before you started the print. Or is it that I have too many twists in my brain because of the strange thinking!!??

Marlin mode - use encoder
Touchscreen mode - use the touchscreen