bigtreetech/BTT-TFT35-E3-V3.0 #58: Thermal runaway protection when using touch screen mode?

markbwalker, 2021-01-18:
Can't seem to determine if the touch screen mode has an option to enable thermal runaway protection.  I think I've exhausted all the menus looking for it with no luck, and I don't see it in the config file either.  Apologies if somehow I missed it.

I could use the Marlin mode since I know it's there, but I have a hitch.  The SD microcard on the SKR Mini E3 V2 failed, so I use the slots on the TFT35 instead.

Running the latest BTT TFT35 files as of 1/18/2021.

oldman4U, 2021-01-18:
Hi. I would recommend to close this ticket here and ask the same questions in the firmware repository.

markbwalker, 2021-01-18:
> Hi. I would recommend to close this ticket here and ask the same questions in the firmware repository.

Thanks - makes sense.

closed 2021-01-18 by markbwalker

markbwalker, 2021-01-18:
Moved to TFT firmware section.