bigtreetech/BTT-TFT35-E3-V3.0 #59: Pinot diagram needs work

xeddog, 2021-01-23:
The pinout diagram appears to have all of the pins, but the display image in the front and not the back where the pins are.  Are the pins backwards, or just the image?

radek8, 2021-01-23:
Yes, it is illogical.
The image is on the side of the display, not the connectors.
The description corresponds to the picture.

radek8, 2021-01-23:
[image]

tkhub, 2021-12-13:
[image]
I have confirmed pin assign by circuit tester.
The manual explains top (flont) view.
(NOT BUTTOM VIEW)

closed 2023-03-25 by bigtreetech