bigtreetech/BTT-TFT35-E3-V3.0 #61: Printer halted. Kill() called!

Grenda1, 2021-01-30:
I recently installed the SKR1.4 turbo with the TFT35 E3 touchscreen. 

During some prints the LCD display shows the the following error 'Printer halted. Kill() called!' and the print stops. 

Everything else appears to be working flawlessly.

Please help.

radek8, 2021-01-31:
Check the thermal protection settings in Marlin

oldman4U, 2021-02-04:
Please answer radek8

Grenda1, 2021-02-04:
Hi Radek8.

Thank you for your email. I've changed some thermal project settings but
still getting the same result.

Looks like the heat bed connections may be intermittently faulting. I am in
the process of checking all connections. Sometimes, the voltage on the SKR
1.4 Turbo mainboard for heater out is not reading 12v when bed heating is
called.

Regards,
Dean.

radek8, 2021-02-04:
At 12 V, too large currents flow.
There may be a problem with the connectors or the fuse. The switching transistor can also overheat.
It would be better to use 24V if possible.
The currents could be half the current.

Grenda1, 2021-02-09:
Hi Radek,

I think I've narrowed it down to the HB 12v output that goes to the mosfet.

When I turn on the printer and set the heatbed to 60ºC, the voltage across
these terminals reads close to 0V. Note this issue is intermittent, and
when the bed is working, these values are 12v, and the mosfet output
voltages are 12v.

Would this indicate an issue with the SKR 1.4 Turbo board???

Thanks in advance,
Dean.

radek8, 2021-02-09:
The problem will be on SKR
If you can, return it to the seller.
If this is not possible, use an external mofset to power the bed

Grenda1, 2021-02-09:
Thanks for the fast response Radek.

The output from the SKR Heated Bed (from the  wsk220n04 ), then feeds into
the external control of the CR10 Heated Bed Mosfet (irfp064n).

Does this seem right to you?

Thanks in advance.

Dean.

radek8, 2021-02-09:
This is the current connection, or do you want to connect it like this?

Grenda1, 2021-02-09:
Currently connected like this

radek8, 2021-02-09:
By connecting an external Mofset, you have the option of fitting better cooling and you do not risk damaging the SKR board.
If one of the Mofset transistors is defective, you can bypass it.
Install a heat sink on a working Mofset.
220W at 12 V, the Mofset is loaded with 18A, that's a lot.
Which mofset are you defective? external or integrated on SKR?

[image]

Grenda1, 2021-02-09:
Please see attached photo. The external mosfet is the original from the
CR10S.

The mosfet that appears defective is the internal SKR HB mosfet. How would
I go about bypassing this?

Heatsink seems like a good idea. Not sure if the board is already defective
though. I can get the hotbed to turn on intermittently. I've put a temp gun
on the mosfet chip and it appears to be running cool when it is operating
correctly.

radek8, 2021-02-10:
Then it's easy.
You can bypass the faulty mosfet and connect pin P2.05 from the processor directly to the external mosfet.

radek8, 2021-02-10:
[image]

radek8, 2021-02-10:
Even better, so you don't have to solder, reconfigure another pin that supports PWM :-)

Grenda1, 2021-02-11:
Hi Radek,

I'm lost as to what the solution is. Also, my TFT35 screen is now turning
off after around 5 seconds of power. I've tried removing cables, and still
not having any luck. I've reflashed it with a working firmware, and it gets
to about 35% (5 seconds in) and then goes black.

I'd like to just replace the mainboard and screen. I purchased originally
from Aliexpress in Oct 2020. Am I still covered under warranty?

Thanks in advance.

radek8, 2021-02-11:
I don't know how you got it with the guarantee. If you are convinced of defective goods, open the dispute.
Podle toho co píšeš, že občas netopí postel, displej se vypíná a podobně.  
Does this indicate a faulty power supply, or a problem with the cabling, possibly a loose fuse?

oldman4U, 2021-02-11:
Take a USB cable and cut it. Remove the TFT from the printer and Connect only the red and black wire to 5V and GND. Put the USB connector in a USB outlet to power up the TFT. See it it works then.

Grenda1, 2021-02-11:
Thanks oldman4u. I managed to plug in only the exp1 connector on the skr
1.4 board which was enough to power it, and reflash the firmware.

As soon as I connected the second exp 2 connector, the screen would go
black.

I'm assuming this indicates the screen is okay, and there is an issue with
the skr board??

@Radek, where do I go to open a dispute for a replacement board? Thanks.

radek8, 2021-02-11:
What you're describing looks like poorly connected cables.
Send a photo of the connection

oldman4U, 2021-02-15:
Please help the community and close the ticket once you do not need it anymore. Thank you

Grenda1, 2021-02-15:
Hi Oldman4U,

I'm still having the screen issue, I will try and take photos for you soon.

Seem to have worked out why the hotbed is shutting down. The pins on the
back of the control box were making a poor connection, causing intermittent
connection issues, and the plugs/wires to get quite hot.

I am yet to reconnect the skr 1.4 board (running oem at the moment).

Grenda1, 2021-02-17:
Hi guys,

Current situation is that when the Skr turbo board is only plugged into
power (12v supply), and the TFT35 screen connected, the screen turns on
(BTT logo), displays icons for a moment, then shuts down.

I also noticed when trying to flash firmware onto the SKR board, it is not
changing the filename of firmware.bin to firmware.cur.

Any suggestions?

radek8, 2021-02-17:
Send photos involved