bigtreetech/BTT-TFT35-E3-V3.0 #63: Display won’t power on when printer is turned on, only if USB plugged into mainboard first!

HeartlessDarkangel, 2021-02-12:
Hey guys, I’m wondering if anybody can help me. 
I’ve had this issue before but it had been fine for quite a while, but recently it has become a steady problem. 

So the problem I’m having is that my display will stay blank and not show anything unless I plug in the USB cable from my computer to the USB of the SKR Mini E3 V2 main board of my ENDER 3 first, it does NOT turn on when I turn on the Main power supply for some weird reason. 

I’ve checked and double checked all the connections and cables and as I said it’s been an intermittent problem but it has become permanent this week. 

What could be causing for the display to only turn on when the USB is plugged in first?
Do I have a defective Mainboard or Display?

Should I just feed 5v power into the back of the TFT35 V3 power port from the main board?
TIA

HeartlessDarkangel, 2021-02-12:
I have plugged in the old display and it powers on when the machine comes on, so I am definitely leaning towards a defective display. I bought this last December from a seller in AliExpress, how would I go about warranty?￼

oldman4U, 2021-02-13:
Most likely the mainboard is not able to deliver enough power for the TFT display.

Please help the community and close the ticket once you do not need it anymore. Thank you

HeartlessDarkangel, 2021-02-13:
> Most likely the mainboard is not able to deliver enough power for the TFT display.
> 
> Please help the community and close the ticket once you do not need it anymore. Thank you 

I'm using it with the SKR Mini E3 v2...

oldman4U, 2021-02-13:
In case it is not broken it should work.

You can test the TFT by removing everything from the TFT and to connect
only 5V and GND. This could be the red and black wire of a USB cable which
you cut off. Once turned on you should be able to do everything with the
TFT except Marlin mode and that you can not control the printer. If this
works you know that the problem is not the TFT itself but a cable or the
mainboard.

HeartlessDarkangel, 2021-02-13:
If either one of them are malfunctioning, how do I go about the warranty on them?
Who do I contact?

oldman4U, 2021-02-13:
The dealer you bought it from.

um 00:14: