bigtreetech/BTT-TFT35-E3-V3.0 #64: BTT-TFT35-E3-V3.0 User Guide points to using BTT-TFT35-V3.0 firmware to update and not the E3-v3.0 firmware

Heratiki, 2021-02-24:
Is this intentional?  If so it would explain why I cannot get my screen do do anything after flashing the newest BTT-TFT35-E3-V3.0 firmware bin file.

oldman4U, 2021-02-25:
The pdf is a mess and needs a complete rewrite. It would be great if you could help the community and write a new one and make a PR. 

Thank you

radek8, 2021-02-27:
The FW for these two screens is almost identical.
The only difference is in the LED under the knob

thisiskeithb, 2021-03-13:
They weren’t separate originally, so the manual would need a new revision.

oldman4U, 2021-03-14:
Keith

Wouldn't it be good to change all documentation to be .md or is there a particular reason why they are pdf's?

TimListon, 2021-03-15:
It would sure make it a lot easier to submit updates to the documentation going forward if it were converted to markup format.

macro312, 2021-08-01:
> The pdf is a mess and needs a complete rewrite. It would be great if you could help the community and write a new one and make a PR.

Seems like BigTreeTech likes to dump inferior products on the community and expect them to do all the work.

oldman4U, 2021-08-01:
@macro312 Yes you are right. 

Please check what "Open Source" means.

This is one of the reasons why it is so cheap and why also you should contribute something USEFULL to this project.