bigtreetech/BTT-TFT35-E3-V3.0 #66: BTT TFT35 E3 V3 problem with RS232 cable

Brutalier, 2021-03-08:
Hi.
I was going to change my LCD screen to TFT in my Ender 3 max, but there is problem with RS232 cable...?
I connect EXP 1 and 2 from TFT to SKR 1.4 turbo and everything works fine. But when I connect that black 5-pin RS232-cable, screen turns to messy, white and you can barely see icons from the screen... When you disconnect cable, everything works! I can change between marlin mode and touch mode...
Is it just in firmware or what...?
Is that RS232-cable really necessary to connect? 

Thanks and sorry... :)

radek8, 2021-03-08:
An RS232 cable is required for communication in touch mode.
Try disconnecting cables EXP1 and EXP2, connect RS232 cable,
Make sure that it is correctly connected and oriented!
What will the display show then?
Can you attach photos?

Brutalier, 2021-03-09:
Last night I powered SKR board with USB and I connect RS232 cable only. And screen works! Then I connect EXP1 and 2, and it still works! Both marlin mode and touch mode. Earlier I powered SKR from enders 24V powersupply... Have I missed something? I will try later this week to connect the SKR ro the printer. I will inform to you what happens.
Thank you for the information.

Brutalier, 2021-03-10:
I found the problem! It was in firmware... I've got TFT port wrong...

closed 2021-03-10 by Brutalier