bigtreetech/BTT-TFT35-E3-V3.0 #69: Encoder bug

craftyham, 2021-03-23:
I have a E3 mini 1.2 & TFT35 E3 V3, running latest version of Marlin bugfix. I've noticed the encoder can control the selected or last moved axis in motion(move) on the touch screen screen mode, when i returned to the main screen the speed percent was increased. The encoder also increases the speed percentage while on the main screen in touch mode like it should if in Marlin mode.

radek8, 2021-03-24:
This is a known issue.
It was solved using cables EXP1 and EXP2.
The problem persists when using the EXP3 cable.
This is your case.

Many tickets have already been opened on this topic.
for example.
https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/57

oldman4U, 2021-03-27:
Please help the community and close the ticket once you do not need it anymore. Thank you

craftyham, 2021-03-27:
> 
> 
> This is a known issue.
> It was solved using cables EXP1 and EXP2.
> The problem persists when using the EXP3 cable.
> This is your case.
> 
> Many tickets have already been opened on this topic.
> for example.
> #57

Thank you for catching me up on the issue. I'll look into in further using the ^57link^

closed 2021-03-27 by craftyham

fiveorders, 2021-04-17:
I didn't want to open up another thread for the same issue, I am using TFT70 and EXP1 and EXP2 cables connected with the Serial cable. Speed changes with the encoder knob when adjusting other things. I have flashed the nightly build Marlin which has been recommended on the other thread. 

Any news about this? If it is not fixed, is it possible to disable the speed alteration feature by the knob? Or is it possible to completely disable the knob on Touch Screen Mode?

oldman4U, 2021-04-17:
You can not use the knob in touchscreen mode.

Using EXP1 and 2 with v27 of the touchscreen firmware this issue should be fixed.

Using EXP3 this issue can not be fixed because the hardware does not allow this.

fiveorders, 2021-04-17:
I downloaded v27 going to try in an hour after the print job. Is the knob completely being disabled on this new firmware? You need the knob button to switch between modes... Thank you

oldman4U, 2021-04-17:
The button should work normally in touchscreen and Marlin mode but does not change speed when used in touchscreen mode. But again, this only works in case your TFT is connected to the mainboard using EXP1 AND 2.

fiveorders, 2021-04-17:
Yes I connected EXP1 EXP2 and the Serial for Touchscreen Mode. Many thank you again.

radek8, 2021-04-17:
As oldman4U writes
- the condition is the current FW in TFT
- connection of cables EXP1 and EXP2
- current Marlin (nightly build)

Marlin (nightly build) provides a write in the pin definition file for your board that deactivates the encoder in touch mode.
You didn't say what kind of board you have with Marlin, so I can't say that your board has the necessary adjustment in the definition of pins.

PS:
You do not only need to switch the encoder to switch modes, but also hold your finger on the display :-)

fiveorders, 2021-04-18:
I use the FLYF407ZG. I have successfully installed the v27. Will check this soon. Many thanks, guys.

radek8, 2021-04-18:
Your board has not defined the necessary modification.
Add the following line to the pins_FLYF407ZG.h file

  #if ENABLED (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
  #define BTN_ENC_EN EXPA1_03_PIN // Detect the presence of the encoder
  #endif

radek8, 2021-04-18:
[image]

radek8, 2021-04-18:
In Marlin you only need to activate 1 display driver:
And this:

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

fiveorders, 2021-04-18:
Will do! Thanks, one question, will the SD card be available both in touch and text mode?

radek8, 2021-04-18:
It depends on the parameter setting in Marlin
#define SDCARD_CONNECTION

When setting up (I prefer)
#define SDCARD_CONNECTION ONBOARD
Only an onboard SD card will be available in Marlin.
Onboard and TFT cards are available in touch mode.

#define SDCARD_CONNECTION LCD
In both Marlin and touch mode, only the SD card on the TFT will be available.

fiveorders, 2021-04-18:
I have defined #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

but cannot find pins_FLYF407ZG.h on my Marlin repo? Should I download it from somewhere

radek8, 2021-04-18:
[image]

fiveorders, 2021-04-18:
Awesome, thank you, now flashing and trying

fiveorders, 2021-04-18:
#define SDCARD_CONNECTION LCD
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

I defined these and got error as below, does not compile

Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2405:4: error: #error "Please select only one LCD controller option."

radek8, 2021-04-18:
And what do you not understand about that mistake? :-)
Comment out the other LCD drivers

fiveorders, 2021-04-18:
//#define REPRAP_DISCOUNT_SMART_CONTROLLER    yes this one:) brain freeze sorry:)

fiveorders, 2021-04-18:
> Your board has not defined the necessary modification.
> Add the following line to the pins_FLYF407ZG.h file
> 
> #if ENABLED (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
> #define BTN_ENC_EN PE7 // Detect the presence of the encoder
> #endif
> 
> Sorry for the mistake, corrected

I added the text below to pins_FLYF407ZG.h because I am using 2004 mode. Still, the speed being altered when I am changing Nozzle temp i.e. 

#if ENABLED (REPRAP_DISCOUNT_SMART_CONTROLLER)
#define BTN_ENC_EN PE7 // Detect the presence of the encoder
#endif

EXP1 and EXP2 and TFT Serial cables are connected. Marlin Nightly build installed.

Any ideas? Thanks

radek8, 2021-04-18:
I provided the ideas, you didn't follow them ...

radek8, 2021-04-18:
Add the following line to the pins_FLYF407ZG.h file
#if ENABLED (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
#define BTN_ENC_EN PE7 // Detect the presence of the encoder
#endif

In Marlin you only need to activate 1 display driver:
And this:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

set in the config.ini file:
marlin_type:0

fiveorders, 2021-04-18:
So this fix won't work with 2004 LCD mode? It needs to be 12864 mode. Am I correct?

radek8, 2021-04-18:
yes

radek8, 2021-04-18:
Display 2004 - HD47780
uses Pin D7 for communication and therefore cannot be used to turn off the encoder.

fiveorders, 2021-04-18:
Probably you are questioning why would someone uses 2004 if there is 12864, there is an explanation for sure. Many thank you for your help. Please send your contact information to venedikli at gmail com and also if you have oldman4u contact information please send that as well. I will add you guys to a list for some perks...👍

radek8, 2021-04-19:
I don't know what the benefits of LCD 2004 are.
If you insist on LCD 2004 and don't need a beeper.
Use the driver for LCD 2004 in Marlin. Deactivate sound in Marlin. Use the beeper control pin to turn off the encoder.

radek8, 2021-04-19:
I would have one less comfortable solution, but it allows any display, including LCD 2004.
The speed only changes when Marlin is on the status screen and you turn the knob.
In Marlin, turn off automatic return to the status screen.
After turning on the printer in touch mode, click the knob once.
Marlin goes to the menu and stops responding to turning the knob.
Now you can turn the knob, but be careful, don't try to click :-)
You use the long touch on the display to switch the mode.

fiveorders, 2021-04-19:
Okay many thanks, I will consider the previous option without sound. Or stay with the 12864

oldman4U, 2021-04-19:
Or simply use the touchscreen in touchscreen mode and the encoder in Marlin
mode.

Or, replace the mainboard😁

Happy printing

radek8, 2021-04-20:
@fiveorders 
Were you successful? Did you manage to deactivate the speed change by turning the knob in touch mode?

fiveorders, 2021-04-20:
@radek8 yes many thank you. I moved to 12864 mode for this purpose. However I have a worser situation after upgrading to v27 on TFT70 and Marlin Nightly Build 2.0. I am having little blobs during the print due to very short pauses of the extrduer.... 

Searched for it and saw that I am not the only one. I think I sadly had to give up on TFT70 and move to 2004 or 12864

fiveorders, 2021-04-20:
You guys didn't send us your contact information, did you? @radek8 @oldman4U

radek8, 2021-04-20:
What contact information do you need?

fiveorders, 2021-04-20:
> What contact information do you need?

your emails... 👍

oldman4U, 2021-04-20:
We are available exclusively through this repository 🤓😜

fiveorders, 2021-04-20:
Alright guys, not for private support, I just wanted to do some favours to you as you helped me a lot here... thanks anyway

oldman4U, 2021-04-20:
You are very welcome. radek is the hero in this case. 👍🏻