bigtreetech/BTT-TFT35-E3-V3.0 #70: Super buggy SD card slot

moebis, 2021-04-18:
This thing has given me so many problems and errors that I'm wondering why I even changed the stock Ender 3 screen, as I only use this thing in Marlin mode now. After flashing everything works fine, but on the next reboot it's saying something about illegal firmware or something and I have to make sure I remove the card after powering down or I have to refresh all over again. Other then a pretty touch screen, I don't get it. I use octoprint, I get no added benefit with this thing, except the nice orange color in marlin mode to match my printer. It seems BTT has a lot of problems implementing SD card slots, even their new kickstarter printer has these problems from the YouTube reviews I've seen with Angus and others. The SKR Mini E3 V2.0? Perfect in every way, love that thing. Wish they put the same forethought into the TFT35. Wish I hadn't bought this in a kit so I could return the TFT35.

radek8, 2021-04-19:
You're cursing at pretty good HW.
And what is your problem with TFT? If I omit the problem with the SD card, which will be enough to format, do you have another problem?

moebis, 2021-04-19:
@radek8 Nobody is cursing. I praised the SKR mini E3 2.0 which is great. My problem is just that, with the TFT, the SD card and how I'm now stuck using it in Marlin mode. Oh I forgot to mention that the touch interface too crashes often when printing from octoprint, so I basically just keep it in Marlin mode. I could have just kept the stock display. Chill.

radek8, 2021-04-19:
Reporting an illegal application could resolve SD card formatting

radek8, 2021-04-19:
Of course, it's a matter of opinion.
I don't use octoprint and the touch screen is comfortable and great for me. I practically don't use Marlin mode anymore.

moebis, 2021-04-19:
It has been reported. Apparently others have the same problem when using a microSD to SD adaptor, which is probably most of us. I can't even remember the last time I used a full sized SD. Even on my older MacBook I would put a microSD in the adaptor and use that, never had a problem. iMac? no problem. I keep formatting the SD cards, putting the firmware.bin on it reloading on the TFT, and if I'm careful and turn off the machine afterwards and remove the SD (never to use it again) then it will boot into the touch interface, but then the touch interface will crash, start throwing errors rapidly over the screen or stop reporting the printer status. Marlin mode works great though. Oh well, you live and learn. Should have realized I didn't really need a fancy color touch screen since I do everything in octoprint anyways. I was briefly entertaining using the wifi module on the TFT and freeing up my raspberry pi, but after this experience I think I'll stay away. Plus I just love everything I can do in octoprint, especially remote monitoring, exact print estimate times, arc welder, etc.

radek8, 2021-04-19:
I'm using a micro SD with an adapter and I have no problem.
Do you have a MacBook? That explains it :-)
Several users here already had the same problem with reporting - illegal applications and all of them formatted and copied the card in their MacBook ...

radek8, 2021-04-20:
see
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1860

moebis, 2021-04-21:
Thank you. Yes I know, but this is still a problem. It's not the Mac's fault. I have numerous devices that use FAT formatted SD cards from my Mac (drone, other sd slot on 3D printers, camera's, etc). Why is the TFT the only one that has an issue. It's a firmware bug they need to fix.

radek8, 2021-04-21:
Try FAT32 format