bigtreetech/BTT-TFT35-E3-V3.0 #71: Touch mode and USB connection issues...

tmwhite86, 2021-05-14:
Hi everybody! 

I have recently installed the TFT35 E3 V3.0.
When the screen is in marlin mode I can still control the printer from my PC.
As soon as I switch to touch mode my printer becomes unresponsive from my PC.
Is there a way to get them to work at the same time of is this how the screen has been designed to work?

My current board is a MKS Gen v1.4.

Thank you in advance for any assistance.

radek8, 2021-05-14:
Hello
The problem is not in the TFT but in the MKS GEN V1.4 board
Both the USB port and the TFT connector share the same serial port.
Only one device can communicate at a time.
Where did you connect TFT? To the AUX-1 connector?
Aux -1 and USB use serial port0

Use the J15 WIFI connector that uses serial port2
Pay attention to the order of the pins. Edit the Marlin configuration.

radek8, 2021-05-14:
Or you can buy a WIFI module and connect the PC via WIFI.

tmwhite86, 2021-05-15:
> Hello
> The problem is not in the TFT but in the MKS GEN V1.4 board
> Both the USB port and the TFT connector share the same serial port.
> Only one device can communicate at a time.
> Where did you connect [...]

Hello.
Yes I connected the tft to the AUX 1 port and the USB to serial 0.
I have tried googling a work around but you have confirmed what I was hoping wasn't true.
The ideal fix would be to buy a compatible board with tft port but I begrudge replacing a board that works. 
I suppose it's not a huge inconvenience as long as I keep it on marlin mode when using octoprint or PC.
I'm sure eventually my board will "break" which will give me a reason to replace it.
I will look into WiFi if possible but from what I have seen so far it wouldn't be as good. 

Thank you for your reply and advice.

radek8, 2021-05-15:
TFT connection to a connector labeled WiFi
In Marlin defines serial connector 2

radek8, 2021-05-17:
If you have received your answer and you have nothing to add, you can close the ticket.
Thank you

closed 2021-05-17 by tmwhite86