bigtreetech/BTT-TFT35-E3-V3.0 #72: Issue: Dialog Text not legible

LTSpenny, 2021-06-19:
[image]
 
I have a few screens and pop ups that have text that reads like the above picture. I used the standard configuration file and TFT35_V3.0_E3.27.x.bin 

any way to fix it? 

Along with this, I tried deleting a few of the "pre set" configurations for other materials I will most likely never use and that screen ended up looking like the above. Thank you for any assistance!

Printer im hooked up to is a Creality Ender Max with a SKR E3 Turbo board running Marlin Bug Fix 2.0.

radek8, 2021-06-19:
The problem has already been fixed. Use the current FW

LTSpenny, 2021-06-19:
Hi sorry im new to this. I thought e3.27.bin was the current. Do you mean just recompile from source? Thank you for taking the time to reply!

LTSpenny, 2021-06-19:
Edit- sorry figured it out just re-downloaded latest files. Thanks for the help!

closed 2021-06-19 by LTSpenny