bigtreetech/BTT-TFT35-E3-V3.0 #74: LCD black screen when powered on with SD Card inserted

robheffo79, 2021-11-06:
Hi,

When I power on my printer with an SD Card inserted into the TFT35's SD Card socket it just sits there with a black screen and does nothing.

In order to get the "Bigtreetech" logo to appear then go into the UI I need to remove the SD Card an power-cycle the machine.

How can I resolve this issue please?

radek8, 2021-11-09:
Use an SD card less than or equal to 8GB.
Format the card to FAT32.
Update TFT to the latest FW.

robheffo79, 2021-11-10:
The SD Card is a 64GB model, but it's partitioned and formatted to a 1GB FAT32 partition. I'll get a new SD Card tomorrow and see how it goes. Thanks

radek8, 2021-11-10:
A 64G SD card formatted to 1G is not the same as a 1G SD card.
SD cards are divided by type into 3 groups according to the standard.
SD - size up to 4G
SDHC -size 2G -32G
SDXC - size 32G -2T
Therefore, some readers may have trouble reading newer types of SD cards.