bigtreetech/BTT-TFT35-E3-V3.0 #75: marlin mode runs in the background

lupleime, 2021-12-30:
when in touch mode, marlin mode keeps running in the background and still does stuff like autohome

[https://photos.app.goo.gl/yoYsAdYmKDbeMRg59](url)

TFT35 E3 V2.0

radek8, 2021-12-30:
The display is connected to the Marlin board with one 10-pin EXP3 cable
Therefore, this is the standard behavior when using EXP3.
I've explained it here several times.
If we had a board with 2 cables EXP 1 and EXP2. the problem is addressed.
When using the EXP3 cable, there is no additional free pin to disable the encoder in marlin mode.
The encoder is still physically connected to the Marlin board and still reacts to movement and clicking on the encoder when switching to Marin mode.

You must not click on the encoder in Marlin mode!
Use a long touch on the screen to switch modes.
Also, rotating the encoder will make a problem. Print speeds are changing.
However, this can be handled by turning off automatic return to the status screen in Marlin mode. Print speed changes only on the Marlin status screen.
The last best but most difficult solution is Marlin's own compilation. Stretch another wire between the Marlin board and the TFT and adjust the pin set. Then Marlin would not be affected when switching to touch mode.

radek8, 2021-12-30:
https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/57