bigtreetech/BTT-TFT35-E3-V3.0 #76: [FR] Equal length cables on the TFT35 E3 V3

V1EngineeringInc, 2022-01-07:
[image]

Hi Team, I have asked my contact at your company, Mike, about this and he said it makes no sense. So I am making this post to try and clarify.

The cables go to the same place, the screen, when you have ribbon cables of different lengths it makes cable management very difficult. On top of that the shorter cables do not allow for the screen to be very far away from the printer or other machine. So if you mount the control board on the side or back, you can not access the screen. Ideally It would be nice to have them a bit longer as well

I hope this is an inexpensive way to vastly improve the usability of your product! Sure to benefit customer satisfaction. If you search for "TFT35 Extension Cables" you will see we are not the only user base looking for this feature.

Thanks for your time.

radek8, 2022-01-15:
You're right. Unfortunately, these are the little things that BTT has a bad reputation for.
They make a quality display at a good price and they put cables that are completely unsuitable. The serial cable is long but not shielded, the EXP cables are short. So not a single cable can be used ....

V1EngineeringInc, 2022-01-15:
Well hopefully someone reads this and can see such a simple thing makes a good product excellent, and that the details like this matter. They make them for me custom, so it is easy enough to have done.

radek8, 2022-01-15:
I'm afraid no one at BTT reads this site. The answer to any question is zero. Only users who are looking for help with their problem respond. BTT sometimes they react to FB

V1EngineeringInc, 2022-01-15:
That is a bummer. Who is the person handling the PR's for the firmware? Perhaps we can convince them to pass along the info.

radek8, 2022-01-15:
PR largely creates a community of users who share their experiences and ideas with others.
PR then merges the user with the name bigtreetech or Msq001
Bigtreetech does not respond to requests for help with HW or HW complaints, in the best case it will tell you to contact your dealer.
Communication is very weak or non-existent.

radek8, 2022-01-15:
Use the EXP1 cable, the one that was at the printer, the other one you make. Discard a serial cable that is of sufficient length and make your own from a shielded cable. I used a cable from an old USB mouse
[image]
.

V1EngineeringInc, 2022-01-15:
Wow, that is eye opening. That is consistent with me buying direct and asking them the same questions and basically getting blown off. 

Bummer, the hardware and firmware do have some issues, but all seem to be fairly simple. I really do hope they start to pay a little more attention to the good products. They seem to be releasing new stuff at a screaming pace, taking care of the products they have should be a bit higher priority. I am willing to help with instructions and firmware, but unfortunately can not help with hardware. This could be a great product, if the price came down a bit and the customization was not necessary (or stays free).

Unfortunately, I get these buy the hundred per order, so fabricating cables is not feasible.

radek8, 2022-01-15:
I won't help you with that. :-(
If you buy hundreds of pieces directly, you have a better bargaining position.
You can buy cables elsewhere, but this will reduce your profit or make the product more expensive.

radek8, 2022-01-15:
For example, the SKR 2 board
Personally, I call her a big mess.
It is still being sold for the third time and no one will help unhappy users unless they help themselves. If you order it, you don't know what will come to you until you have it in your hand ...

V1EngineeringInc, 2022-01-15:
I have not ventured too far into the electronic manufacturing side of things, but I do not really understand why they make so many models. Most of the boards have plenty of CPU power to spare and plenty of extra ports. 

On the consumer side of things, I would much prefer the economy of scale to kick in and drop the prices a bit. Writing new instructions and tracking down firmware bugs on new boards every month or so is time-consuming. 

One of the other boards we have been using for five plus years still has plenty of power and when it did undergo a hardware revision they kept the firmware side of things the same. No changes needed to happen on my end, other than things got cheaper, more robust and most importantly easier to use.